phoenix-rtos/phoenix-rtos-project #639: Fscanf /vfscnaf problem with setting file position indicator

maska989, 2023-03-10:
* Using fscanf with specifier %c shall move the position indicator one place further but in this case, this function tries to pass over all characters in the file.

* Tested on Host-generic-pc and Ia32-generic-qemu.

* To reproduce issue copy following code:

``` c
#include <stdio.h>
int main(void)
{
	char c[1];
	FILE *rd;

	rd = fopen("test", "w+");

	fputs("abc", rd);

	fseek(rd, 0, SEEK_SET);
	printf("\nposi before = %ld\n", ftell(rd));

	fscanf(rd, "%c", c);
	printf("\nposi after = %ld\n", ftell(rd));

	fclose(rd);
}
```

* Outcome on Host-generic-pc and ia32-generic-qemu:
[image]

gerard5, 2023-03-14:
@maska989 I added a fix, please verify if it works in your test cases.

maska989, 2023-03-14:
@gerard5 Works like a charm.

maska989, 2023-04-11:
https://github.com/phoenix-rtos/phoenix-rtos-tests/blob/c13ef424dff7c1f0f88f02e4a67104f6d9ed5d22/libc/stdio_scanf.c#L3330-L3333

closed 2023-04-25 by gerard5