phoenix-rtos/phoenix-rtos-project #652: Scanf doesn't support longer floating-point numbers

maska989, 2023-03-13:
**Problem occurs when floating-point numbers are used with high precision after coma**

Testesd between ia32-generic and host-generic

* Code to reproduction:

``` c
#include <stdio.h>
#include <stdlib.h>

int main(void)
{
	float f;
	sscanf("0.0000000000000000000000000000000011755", "%f", &f);
	printf("\n%.40f => 40 scanf", f);
	printf("\n%.40f => 40 strtof\n",
		strtof("0.0000000000000000000000000000000011755", NULL));
}
```

* Outcome: 
[image]

But after further investigation, @niewim19 found a solution.
In scanf.c we need to change line 110
* From this:
[image]

* To this (value to test):
[image]

And outcome of what we got matched host-generic

[image]

niewim19, 2023-03-13:
Well... :) I have just found that we have a buffer of length 32 that is insufficient to hold the FLT_MIN (given it is 4 byte long) in decimal representation. This final buffer length should assessed as if we approach long double this buffer will bloat very much.

gerard5, 2023-03-14:
Thanks @niewim19 for the tip. Of course, the buffer needs to be enlarged, but ***not at the expense of the stack*** (which is precious on small targets). Perhaps one solution might be to increase the `CCL` buffer which is already `malloc'ated` and share/use half of the space available for buffering purposes, another way might be to grow with `realloc` when it becomes necessary during format parsing. I'll try to figure it out and provide a fix.

maska989, 2023-04-11:
https://github.com/phoenix-rtos/phoenix-rtos-tests/blob/c13ef424dff7c1f0f88f02e4a67104f6d9ed5d22/libc/stdio_scanf.c#L2641-L2647

Darchiv, 2024-11-13:
Fixed by https://github.com/phoenix-rtos/libphoenix/pull/370

closed 2024-11-13 by Darchiv