phoenix-rtos/phoenix-rtos-project #657: Strtod doesn't handle hex numbers floats

maska989, 2023-03-14:
**Problem occurs when hex input float is used *

Tested between ia32-generic and host-generic
* Code to reproduction:

``` c
#include <stdio.h>
#include <stdlib.h>
#include <limits.h>

int main(void)
{
	float f = 3.40282347e+38F;

	printf("printf =>\n%f || %a || %e || %g\n", f, f, f, f);

	printf("\nstrtod %%a=>\n%f ||", strtod("0x1.fffffe0000000p+127", NULL));
	printf(" %a ||", strtod("0x1.fffffe0000000p+127", NULL));
	printf(" %e ||", strtod("0x1.fffffe0000000p+127", NULL));
	printf(" %g\n", strtod("0x1.fffffe0000000p+127", NULL));

	printf("\nstrtod %%e=>\n%f ||", strtod("3.402823e+38", NULL));
	printf(" %a ||", strtod("3.402823e+38", NULL));
	printf(" %e ||", strtod("3.402823e+38", NULL));
	printf(" %g\n", strtod("3.402823e+38", NULL));

	printf("\nstrtod %%g=>\n%f ||", strtod("3.40282e+38", NULL));
	printf(" %a ||", strtod("3.40282e+38", NULL));
	printf(" %e ||", strtod("3.40282e+38", NULL));
	printf(" %g\n", strtod("3.40282e+38", NULL));
}
```

* Outcome: 
[image]

Darchiv, 2024-07-05:
Applies to `strtod()`, `strtof()`, `strtold()`

Darchiv, 2024-11-13:
Fixed by https://github.com/phoenix-rtos/libphoenix/pull/370

closed 2024-11-13 by Darchiv