phoenix-rtos/phoenix-rtos-project #677: Scanf doesn't read 0x0 as null pointer

maska989, 2023-03-30:
* When using scanf to read 0x0 value on host-generic we get (nil), but on our system scanf does not recognize it as (nil).

 * Code to reproduce issue:
``` c
#include <stdio.h>

int main(void)
{
	void *ptr = (void *)0x1;

	printf("ret = %d\n", sscanf("(nil)", "%p", &ptr));
	printf("ptr = %p\n", ptr);
}
``` 

 * Output on ia32-generic-qemu and host-generic-qemu:
[image]

maska989, 2023-04-11:
https://github.com/phoenix-rtos/phoenix-rtos-tests/blob/c13ef424dff7c1f0f88f02e4a67104f6d9ed5d22/libc/stdio_scanf.c#L3548

closed 2023-09-28 by agkaminski

badochov, 2023-10-02:
@maska989 is the mentioned test reenabled?

maska989, 2023-10-02:
> @maska989 is the mentioned test reenabled?

https://github.com/phoenix-rtos/phoenix-rtos-tests/pull/265
There is PR, sorry for late update