phoenix-rtos/phoenix-rtos-project #679: Problem with `fscanf` and `vfscanf` using brackets and `\n`

maska989, 2023-03-31:
* Problem occurs only in cases when brackets are used with scanf operating on files (fscanf/vfscanf).
 * In the progress of creating tests for square bracket format there couple of possibilities for interrupting this read, but there was found a difference between our system and host generic in placing `\n` in the text. 
 * After a deeper investigation scanf, found that fscanf and vfscanf read `\n` as the end of the file.

 * Code to reproduction:
``` c
#include <stdio.h>

int main(void)
{
	FILE *fd;
	char buff[12] = "abc\nabc";
	char exp[12] = { 0 };
	char c1, c2, c3, c4, c5, c6, c7;
	int sz;

	fd = fopen("test", "w+");
	fprintf(fd, "%s", buff);
	rewind(fd);

	fseek(fd, 0L, SEEK_END);
	sz = ftell(fd);
	rewind(fd);

	printf("\nFile contetn length = %d\nScanf = %d =>", sz, fscanf(fd, "%[^x]", exp));
	printf("%s<=\n\n", exp);
	rewind(fd);
	printf("\nFile contetn length = %d\nScanf = %d =>", sz, fscanf(fd, "%c%c%c%c%c%c%c", &c1, &c2, &c3, &c4, &c5, &c6, &c7));
	printf("%c%c%c%c%c%c%c<=\n\n", c1, c2, c3, c4, c5, c6, c7);
}
```
 * Code output on ia32-generic-qemu and host-geneic-pc
[image]

maska989, 2023-04-11:
https://github.com/phoenix-rtos/phoenix-rtos-tests/blob/c13ef424dff7c1f0f88f02e4a67104f6d9ed5d22/libc/stdio_scanf.c#L3972

jmaksymowicz, 2024-07-19:
This happens because the parser for `scanf`, `fscanf`, `vscanf` and `vfscanf` loads only one line of the file using `getline`: https://github.com/phoenix-rtos/libphoenix/blob/ffca1e635e8146d77b5694077f5283cb88f0154b/stdio/scanf.c#L665

closed 2026-10-20 by ziemleszcz