phoenix-rtos/phoenix-rtos-project #682: stat() completion successful and set wrong errno, when a component of the path prefix is not a directory

mateusz-bloch, 2023-04-05:
stat() completion successful and set wrong errno, when a component of the path is existing regular file, not a directory

Tested between ia32-generic and host-generic

Code to reproduction:
```
#include <stdio.h>
#include <string.h>
#include <fcntl.h>
#include <sys/stat.h>
#include <sys/types.h>
#include <unistd.h>
#include <errno.h>

int main(void)
{
	struct stat buf;

	FILE *fp = fopen("testfile.txt", "w");
	fclose(fp);

	errno = 0;
	stat("testfile.txt/", &buf);
	printf("Regular File: %d\n",S_ISREG(buf.st_mode));
 	printf("Directory: %d\n",S_ISDIR(buf.st_mode));
	printf("Errno: %d, %s\n", errno, strerror(errno));

	remove("testfile.txt");

	return 0;
}
```

Outcome from ia32-generic:
[image]

Outcome from host-generic:
[image]

Documentation:
[image]
[image]

nalajcie, 2023-04-05:
Please note that:
- when function exits with error, output parameter might be left unchanged - IMHO You shouldn't test it unless it's explicitly stated it's also set  on error paths
- most of our FS-related functions are implemented inside particular filesystem - it's worth to check also how the function behaves on jffs2 and dummyfs (I believe it's possible to test on zynq-qemu as rootfs is `jffs2` there and `/dev` is `dummyfs` ; currently it's not possible to test other FSes on `ia32` due to phoenix-rtos/phoenix-rtos-project#114)
-  due to above - please state **on which filesystems the bug is present** (preferably all fliesystems in single issue)

jmaksymowicz, 2024-02-16:
I managed to track down the bug to the `resolve_path` function in libphoenix. According to [Open Group](https://pubs.opengroup.org/onlinepubs/9699919799/) it should set `errno` to `ENOTDIR` when:
> [...] the file_name argument contains at least one non-\<slash\> character and ends with one or more trailing \<slash\> characters and the last pathname component names an existing file that is neither a directory nor a symbolic link to a directory.

Instead, the function simply ignores trailing slashes, so it cannot behave correctly.

closed 2026-10-21 by jmaksymowicz