phoenix-rtos/phoenix-rtos-project #717: armv7a9-zynq7000-qemu: couldn't find a prompt

damianloew, 2023-05-04:
[image]

It may be seen during CI test campaigns runs, there is an example:
https://github.com/phoenix-rtos/plo/actions/runs/4881702942/jobs/8710956270?pr=245

phoenix-rtos-project version:
c4c322a75b1333f4797fad184c313dd2feebee64

damianloew, 2023-06-01:
Probably related issue:
[image]

Link from CI:
https://github.com/phoenix-rtos/phoenix-rtos-devices/actions/runs/5143555467/jobs/9258759304?pr=379

phoenix-rtos-project version:
46287a06bc0068842b87f2a8f88f5211f1ca4de9

damianloew, 2023-06-09:
It may also happen this way:
[image]

Note that the order of starting pid's is unusual.

Link from CI:
https://github.com/phoenix-rtos/phoenix-rtos-tests/actions/runs/5200849402/jobs/9420382791

phoenix-rtos-version:
0414b09db9f4cd17ad8cc84b778ebcb2518f722b

nalajcie, 2023-06-19:
- in `psh` the Exception happens in `vfork` (addr2line `lr` points to `pshapp.c:882`)
- in `linuxrc` happens in `fork`/`vfork` (addr2line `lr` points to `init/init.c:479` but I don't have exact unstripped binary)

badochov, 2023-10-02:
Probably fixed by https://github.com/phoenix-rtos/phoenix-rtos-kernel/pull/428