phoenix-rtos/phoenix-rtos-project #718: Using /bin/rm on device throws EINVAL, but deletes it anyway

astalke, 2023-05-04:
Using /bin/rm on a device file throws **EINVAL**, but deletes it anyway.

[image]

Apparently, the expected behavior is to delete a device file without throwing an error.

nalajcie, 2023-05-04:
IMHO deleting some devices should not be possible because there is no safe way of recreating them.

artur-augustyniak, 2023-07-18:
Disclaimer, sorry if I'm using not accurate terminology, started messing around with the code just recently.
Disclaimer 2, tested this only on a32-generic-qemu target, but all the code involved seems to be platform agnostic.

IMO this inconsistency is related to blurry ownership or precedence definition between servers sharing objects linkage.

Basically this is what is happening  when you try to delete one of the device files created and linked to it's own port by posix server, namely those declared here:
https://github.com/phoenix-rtos/phoenix-rtos-posixsrv/blob/master/special.c.
(notice that other device files can be deleted without EINVAL).

You will get mtUnlink message delivered to this posixsrv instance, but... Because those file are also linked to more basic atasrv, this msg will be delivered to atasrv first:
https://github.com/phoenix-rtos/phoenix-rtos-kernel/blob/fa9d23f38657f4040863eb1808ff830cf7bd8641/posix/posix.c#L982

As far as i understood the code, port numbers are assigned more or less incrementally in order with  servers starting and requesting them. In a case of default build/config atasrv will (obviously) have precedence over posixsrv and will be happy  to unlink whatever it was asked for.

Then, in case of /dev/* files  created by posixsrv,  proc_lookup will find also port of the posixsrv instance and continue with full unlink path there.

In effect, when posixsrv is trying to dispatch unlink message, it hits nullcheck for object with given id (removed by atasrv), thus returning EINVAL:

https://github.com/phoenix-rtos/phoenix-rtos-posixsrv/blob/5c8501120ffb3a2f4aee4f579988c04c80aa019d/posixsrv.c#L335

artur-augustyniak, 2023-07-18:
On the other hand, taking architecture considerations into account, I don't now if it should be considered as bug, maybe just a bug in posixsrv. If you assume that you are live in a multi server environment - you should be prepared for objects being nulled, and take action only if reference is not null.

agkaminski, 2023-07-19:
Hi, thanks for your input on this matter. IMHO this is caused by an incomplete API implementation by the device driver - in case of `/dev/null` posixsrv, to be exact lack of `mtUnlink` implementation. Being able to recreate e.g. `/dev/null` after it's removal is a separate issue. IMHO there's no valid reason to forbid deletion of special files in general because of difficulty of recreating them.

closed 2023-08-11 by agkaminski