phoenix-rtos/phoenix-rtos-project #736: Low power mode is not entered when all threads are blocked without timeout

xvuko, 2023-05-24:
Decision to enter low power mode is based on value returned from [_proc_nextWakeup](https://github.com/phoenix-rtos/phoenix-rtos-kernel/blob/c472a711b2e9001cbafb2cc289c857b56c732a8e/proc/threads.c#L1387). When threads_common.sleeping tree is empty this function returns 0 and prevents low power mode to be entered.

Note that I have not tested this on device yet.

agkaminski, 2023-05-25:
We cannot enter low power mode when we don't know how long to sleep. No sleeping threads is not really a use case (most notably not a use case on neither SKAL nor DEND projects)

xvuko, 2023-05-29:
Yes this does not impact current projects.

This can be a case if all threads are blocked on external interrupts (ie. RTC or GPIO).  In such case we can safely sleep as no thread needs to be scheduled without external interrupt.

jmaksymowicz, 2026-08-21:
The current behavior is unintuitive. Unless there is at least one thread in the system that sleeps for a set time or waits with timeout e.g. on `condWait`, the system will never enter low-power - all peripherals will work as normal. Then when such a thread appears, e.g. application calls `sleep()`, suddenly we go to low-power and now our peripherals and wakeup options are limited.

IMO it would be sensible to sleep indefinitely and wait for external interrupts to wake us up. Otherwise we should at least update documentation to describe how low power modes actually work on Phoenix-RTOS.