phoenix-rtos/phoenix-rtos-project #740: riscv64 qemu/spike To qucik PLO and  HAL Exception: 15 Store/AMO page fault

maska989, 2023-05-30:
Looking at Ci we can find that our system can be built onto riscv64 architecture, that's why I tried to run `psh` tests on riscv64-geneirc-qemu and was greeted by HAL exception. First of all, was that I do something wrong but after researching and doing the same on riscv64-generic-spike with a similar outcome.

- **HAL Exception:**

[image] 

After a couple of reboots I found that `(psh)%` appear to "quickly" (before a full load of psh).

- **Quick (PLO)$:**

 
[image]
 
[image]

gerard5, 2023-06-30:
I've identified that these two (below) commits introduce the problem, if you reverse the changes below riscv64 works fine on master branch   [0494ae4](https://github.com/phoenix-rtos/phoenix-rtos-project/commit/0494ae4300dd195e84696732434e949cad87e6f8) , both `spike` and `qemu-system-riscv64`

* support for errno in thread local storage : https://github.com/phoenix-rtos/libphoenix/commit/079fa4447d26206aee88ea022753c6e28900a67c
* scheduler invoked indirectly via interrupts_dispatch (using threads_timeintr) : https://github.com/phoenix-rtos/phoenix-rtos-kernel/commit/87aca363a3da0f7bd4ffd59723a5fbd375626318

See https://github.com/phoenix-rtos/phoenix-rtos-doc/issues/152

lukileczo, 2023-07-03:
This is problem with destruction of elf file, issue here https://github.com/phoenix-rtos/phoenix-rtos-project/issues/529

agkaminski, 2023-07-05:
I disconnected https://github.com/phoenix-rtos/phoenix-rtos-kernel/pull/413 from this issue, as it needs further investigation, thanks @gerard5

closed 2023-07-21 by lukileczo