phoenix-rtos/phoenix-rtos-project #741: libphoenix: pthread_attr_setstack doesn't work on imxrt106x

adamdebek, 2023-05-31:
Thread stack doesn't change its size on imxrt106x (it works on ia32-generic-qemu), code to reproduction:

```C
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>
#include <pthread.h>
#include <errno.h>

void *thread(void* arg) {
	size_t s = 0;
	pthread_attr_t attr;

	int res = pthread_attr_init(&attr);
	if (res == -1) {
		fprintf(stderr, "Error in pthread_attr_init in thread\n");
	}
	if (pthread_attr_getstacksize(&attr, &s) == -1) {
		fprintf(stderr, "setstack error in thread\n");
	}

	fprintf(stderr, "stacksize in thread: %u\n", s);
}

int main(){

        printf("ello\n");
	size_t stacksize = 4096;
	size_t s = 0;
	pthread_attr_t attr;
	pthread_t tid;
	int res;

	res = pthread_attr_init(&attr);
	if (res == -1) {
		fprintf(stderr, "Error in pthread_attr_init\n");
		return 1;
	}
	if (pthread_attr_setstacksize(&attr, stacksize) == EINVAL) {
		fprintf(stderr, "setstack error\n");
		return 1;
	}

	if (pthread_attr_getstacksize(&attr, &s) == -1) {
		fprintf(stderr, "setstack error\n");
	}

	fprintf(stderr, "stacksize: %u\n", s);

	res = pthread_create(&tid, &attr, thread, NULL);
	if (res != 0) {
		fprintf(stderr, "Couldn't create thread\n");
		return 1;
	}
	pthread_join(tid, NULL);
}

}
```

agkaminski, 2023-05-31:
In `thread` you are creating brand new attribute via `pthread_attr_init` function. Default stack size is 512 on imxrt and 4096 on ia32, so by accident this code works on ia32. This check in `thread` function is invalid (as we're not testing changed attribute, but a new one entirely)

nalajcie, 2023-05-31:
please note that:
* the test is invalid - `pthread_attr_init` always gets default system attributes for creating new threads (not *current thread settings*)
* the test seemingly passess probably because the default stack size for MMU platforms is 4096 (so you're not changing the default value at all)
* AFAIK there is no posix way to get the stack size of the current thread so I don't know if a portable test like that can even be written (there is a GNU extension for pthreads (https://man7.org/linux/man-pages/man3/pthread_getattr_np.3.html) but we don't implement it)