phoenix-rtos/phoenix-rtos-project #745: imxrt106x strange behaviour (toolchain error??)

adamdebek, 2023-06-02:
Defining char array with more than 3 elements causes program to work improperly.
Defining char pointer with string literal works fine, defines don't cause problems neither.
Sample code which works:

```C
#include <stdio.h>
#include <fcntl.h>
#include <unistd.h>

#define PATH "testFile"
#define ABC "DASDASDASDASDASDASDAS"

int main(){
	int fd;
	char *nap = "dasdas";
	char abc1[] = "ab1";
	char abc2[] = "ab2";
	char abc3[] = "ab3";
	char abc4[] = "ab4";
	char abc5[] = "ab5";
	char abc6[] = "ab6";
	char abc7[] = "ab7";
	char abc8[] = "ab8";

    fd = open(PATH, O_RDWR | O_CREAT | O_TRUNC);
	if (fd < 0) {
		fprintf(stderr, "Couldn't open file\n");
		return 1;
	}

	printf("ret: %d\n" , write(fd, "123", 4));

	return 0;
}
```

Output:
[image]

However, when i change just abc8 with 1 more element, write returns -1;

```C
#include <stdio.h>
#include <fcntl.h>
#include <unistd.h>

#define PATH "testFile"
#define ABC "DASDASDASDASDASDASDAS"

int main(){
	int fd;
	char *nap = "dasdas";
	char abc1[] = "ab1";
	char abc2[] = "ab2";
	char abc3[] = "ab3";
	char abc4[] = "ab4";
	char abc5[] = "ab5";
	char abc6[] = "ab6";
	char abc7[] = "ab7";
	char abc8[] = "ab88"; //one more element

    fd = open(PATH, O_RDWR | O_CREAT | O_TRUNC);
	if (fd < 0) {
		fprintf(stderr, "Couldn't open file\n");
		return 1;
	}

	printf("ret: %d\n" , write(fd, "123", 4));

	return 0;
}
```
Output:
[image]

agkaminski, 2023-06-02:
What value of `errno` is after failed `write`?

adamdebek, 2023-06-02:
22

agkaminski, 2023-06-02:
There's a bug in the example - when passing `O_CREAT` to `open` you should include third argument - `mode`.

agkaminski, 2023-06-02:
Should be `fd = open(PATH, O_RDWR | O_CREAT | O_TRUNC, S_IFREG);`

closed 2023-06-02 by damianloew