phoenix-rtos/phoenix-rtos-project #779: Porting to VisionFive2 riscv64 hardware

KenDickey, 2023-07-17:
This is basically an information request.  I need to know more about how to knit together u-boot, OpenSBI, and the Phoenix-RTOS kernel.
My simple attempt was to place the kernel and initrd on a working SD card which boots Debial linux and has working dtb's, but something is amiss.

Under Debian
-----
root@starfive:~# cd /boot
/boot

root@starfive:/boot# ls -F
System.map-5.15.0-starfive*  phoenix-kernel.img*
config-5.15.0-starfive*      rootfs.ext2*
dtbs/                        syspagen*
extlinux/                    uEnv.txt*
initrd.img-5.15.0-starfive*  vmlinuz-5.15.0-starfive*

root@starfive:/boot# lsblk
        mmcblk1p1 179:1    0    2M  0 part 
        mmcblk1p2 179:2    0    4M  0 part 
        mmcblk1p3 179:3    0  100M  0 part /boot
        mmcblk1p4 179:4    0 14.7G  0 part /

----
extlinux.conf contains
---

label 10s
        menu label Phoenix-RTOS
        linux /phoenix-kernel.img[B
        initrd /rootfs.ext2
        fdtdir /dtbs
        append  root=/dev/mmcblk1p3 rw console=tty0 console=ttyS0,115200 debug earlycon rootwait selinux=0

-----
Boot log shows
---

U-Boot menu
1:      Debian GNU/Linux bookworm/sid 5.15.0-starfive
2:      Debian GNU/Linux bookworm/sid 5.15.0-starfive (rescue target)
3:      Phoenix-RTOS
Enter choice: 3
3:      Phoenix-RTOS
Retrieving file: /rootfs.ext2
71657472 bytes read in 3093 ms (22.1 MiB/s)
Retrieving file: /phoenix-kernel.img
379928 bytes read in 23 ms (15.8 MiB/s)
append: root=/dev/mmcblk1p3 rw console=tty0 console=ttyS0,115200 debug earlycon rootwait selinux=0
Retrieving file: /dtbs/starfive/jh7110-visionfive-v2.dtb
50235 bytes read in 11 ms (4.4 MiB/s)
Bad Linux RISCV Image magic!
1:      Debian GNU/Linux bookworm/sid 5.15.0-starfive

-----
Now I note that the VisionFive3/linux SBI is version 1.0, where Phoenix uses version 0.3.

Both VF2/linux and Phoenix build u-boot and OBSI, but I don't know the settings and invarients to hook up the flashed VF2 u-boot and OSBI with the Phoenix kernel.

I have spent a bunch of time reading docs and code but could really use some help here.

Thanks for ideas, suggestions, theory of operation, ...
-KenD

https://github.com/starfive-tech/VisionFive2

KenDickey, 2023-07-17:
More log for context..

[VisionFive2-Phoenix.txt](https://github.com/phoenix-rtos/phoenix-rtos-project/files/12073157/VisionFive2-Phoenix.txt)

agkaminski, 2023-07-19:
Hi, please take a look at a process of image generation for `riscv64` target: https://github.com/phoenix-rtos/phoenix-rtos-project/blob/master/_targets/build.project.riscv64-generic

`OpenSBI` is integrated within the system image, so this image is prepared to be launched as is, without any additional bootloaders. Phoenix-RTOS does not support `U-Boot` at this moment, presently there are no concrete plans for it's support. Also, there's a good chance VisionFive2 support require some work to be done, as there might be hardware or even CPU core differences in comparison to the supported qemu and spike emulated targets.