phoenix-rtos/phoenix-rtos-project #782: <ctype.h> functions should support code inlining

xvuko, 2023-07-18:
[`ctype.h`](https://github.com/phoenix-rtos/libphoenix/blob/d26c40eea4c7b9dce43bb69ff295d6c3e35dc92d/include/ctype.h) contains basic functions such as `isdigit` that generate very small code and are frequently used in loops. This makes inlining highly desirable, but current header does not support this.

agkaminski, 2023-07-18:
It should be ok to define these as `static inline`, opengroup permits that https://pubs.opengroup.org/onlinepubs/009696699/basedefs/ctype.h.html

agkaminski, 2023-08-08:
Can't do, C++ needs symbols:
```
i386-pc-phoenix-ld: /home/praca/toolchain-ia32/i386-pc-phoenix/lib/gcc/i386-pc-phoenix/9.3.0/../../../../i386-pc-phoenix/lib/libstdc++.a(istream-inst.o): in function `std::ctype<char>::is(unsigned int, char) const':
istream-inst.cc:(.text._ZNKSt5ctypeIcE2isEjc[_ZNKSt5ctypeIcE2isEjc]+0xff): undefined reference to `isspace'
i386-pc-phoenix-ld: istream-inst.cc:(.text._ZNKSt5ctypeIcE2isEjc[_ZNKSt5ctypeIcE2isEjc]+0x11c): undefined reference to `isprint'
i386-pc-phoenix-ld: istream-inst.cc:(.text._ZNKSt5ctypeIcE2isEjc[_ZNKSt5ctypeIcE2isEjc]+0x139): undefined reference to `iscntrl'
i386-pc-phoenix-ld: istream-inst.cc:(.text._ZNKSt5ctypeIcE2isEjc[_ZNKSt5ctypeIcE2isEjc]+0x156): undefined reference to `isupper'
i386-pc-phoenix-ld: istream-inst.cc:(.text._ZNKSt5ctypeIcE2isEjc[_ZNKSt5ctypeIcE2isEjc]+0x173): undefined reference to `islower'
i386-pc-phoenix-ld: istream-inst.cc:(.text._ZNKSt5ctypeIcE2isEjc[_ZNKSt5ctypeIcE2isEjc]+0x190): undefined reference to `isalpha'
i386-pc-phoenix-ld: istream-inst.cc:(.text._ZNKSt5ctypeIcE2isEjc[_ZNKSt5ctypeIcE2isEjc]+0x1ca): undefined reference to `ispunct'
i386-pc-phoenix-ld: istream-inst.cc:(.text._ZNKSt5ctypeIcE2isEjc[_ZNKSt5ctypeIcE2isEjc]+0x1e4): undefined reference to `isxdigit'
i386-pc-phoenix-ld: istream-inst.cc:(.text._ZNKSt5ctypeIcE2isEjc[_ZNKSt5ctypeIcE2isEjc]+0x1fe): undefined reference to `isalnum'
i386-pc-phoenix-ld: istream-inst.cc:(.text._ZNKSt5ctypeIcE2isEjc[_ZNKSt5ctypeIcE2isEjc]+0x218): undefined reference to `isgraph'
i386-pc-phoenix-ld: istream-inst.cc:(.text._ZNKSt5ctypeIcE2isEjc[_ZNKSt5ctypeIcE2isEjc]+0x232): undefined reference to `isblank'
i386-pc-phoenix-ld: /home/praca/toolchain-ia32/i386-pc-phoenix/lib/gcc/i386-pc-phoenix/9.3.0/../../../../i386-pc-phoenix/lib/libstdc++.a(ctype_configure_char.o): in function `std::ctype<char>::do_toupper(char) const':
ctype_configure_char.cc:(.text._ZNKSt5ctypeIcE10do_toupperEc+0x15): undefined reference to `toupper'
i386-pc-phoenix-ld: /home/praca/toolchain-ia32/i386-pc-phoenix/lib/gcc/i386-pc-phoenix/9.3.0/../../../../i386-pc-phoenix/lib/libstdc++.a(ctype_configure_char.o): in function `std::ctype<char>::do_toupper(char*, char const*) const':
ctype_configure_char.cc:(.text._ZNKSt5ctypeIcE10do_toupperEPcPKc+0x1b): undefined reference to `toupper'
i386-pc-phoenix-ld: /home/praca/toolchain-ia32/i386-pc-phoenix/lib/gcc/i386-pc-phoenix/9.3.0/../../../../i386-pc-phoenix/lib/libstdc++.a(ctype_configure_char.o): in function `std::ctype<char>::do_tolower(char) const':
ctype_configure_char.cc:(.text._ZNKSt5ctypeIcE10do_tolowerEc+0x15): undefined reference to `tolower'
i386-pc-phoenix-ld: /home/praca/toolchain-ia32/i386-pc-phoenix/lib/gcc/i386-pc-phoenix/9.3.0/../../../../i386-pc-phoenix/lib/libstdc++.a(ctype_configure_char.o): in function `std::ctype<char>::do_tolower(char*, char const*) const':
ctype_configure_char.cc:(.text._ZNKSt5ctypeIcE10do_tolowerEPcPKc+0x1b): undefined reference to `tolower'
make: *** [/home/praca/git/phoenix-rtos-project/phoenix-rtos-build/makes/binary.mk:76: /home/praca/git/phoenix-rtos-project/_build/ia32-generic-qemu/prog/hellocpp] Błąd 1
```

anglov, 2023-08-08:
Maybe it's still worth to consider anyways? 

A quick look on musl shows that they defined both macros and function for that.

https://git.musl-libc.org/cgit/musl/tree/include/ctype.h
https://git.musl-libc.org/cgit/musl/tree/src/ctype/islower.c

Not checked what is in newlib/glibc

I also think if `extern inline` won't do the trick but I didn't play with it but it should generate code both with symbol and inline linking possibility

What libstdc++ do is undefining macros for use symbol version, so it will work with musl approach:

https://gcc.gnu.org/git/?p=gcc.git;a=blob;f=libstdc%2B%2B-v3/include/c_std/cctype;h=af348b416d84a053c51523986ab57de8f61b1810;hb=HEAD

Of course two implementations  require more attention in case of tests so it maybe worth to find if current situation is really an issue.

agkaminski, 2023-08-08:
As it turns out, you need to _just_ rebuild the toolchain for this to work

closed 2023-08-11 by agkaminski