phoenix-rtos/phoenix-rtos-project #808: Lack of controlling terminal functionality

adamdebek, 2023-08-25:
Controlling terminal is part of POSIX standard.

nalajcie, 2023-08-25:
Please be more specific in regards to what is missing. AFAIK we do support all termios functionality (see `libtty` for details)

adamdebek, 2023-08-28:
I meant all staff related to job control, controlling processes and terminal associated with session, but since it isn't required by POSIX I guess this issue has to be closed and phoenix.

closed 2023-08-28 by adamdebek

nalajcie, 2023-08-28:
That's why it's necessary to describe clearly what you meant ;).

AFAIK job control is part of POSIX shell specification, psh is not meant to be posix-compliant. We do support BusyBox ash which is posix-compliant. If I remember correctly the job control support is currently not compiled in (might indeed need some OS support which is missing right now). You may verify it yourself and follow up with more specific issue / feature request.