phoenix-rtos/phoenix-rtos-project #819: Unable to find the RISC-V BIOS (looking for phoenix.osbi)

KenDickey, 2023-09-05:
Did a pull, `build.sh all clean`, but no opensbi build. `build.sh image` unhelpful.
Recent Makefile chamge?

```
> Manjaro:Arm64:~/RISCV/phoenix-rtos-project >>> ./scripts/riscv64-generic-qemu.sh
> WARNING: Image format was not specified for './scripts/../_boot/riscv64-generic-qemu/phoenix.disk' and probing guessed raw.
>          Automatically detecting the format is dangerous for raw images, write operations on block 0 will be restricted.
>          Specify the 'raw' format explicitly to remove the restrictions.
> qemu-system-riscv64: Unable to find the RISC-V BIOS "./scripts/../_boot/riscv64-generic-qemu/phoenix.osbi"
> 
```

gerard5, 2023-09-06:
Confirmed, same error on my side. I have added a [patch](https://github.com/phoenix-rtos/phoenix-rtos-project/pull/820) to fix building the riscv64 image. Please check if it solves the problem.

I built like this (`all` may take long time to build `ports`):
```
TARGET=riscv64-generic-qemu ./phoenix-rtos-build/build.sh clean host core project image
```

KenDickey, 2023-09-06:
Works great.  Thanks much!

BTW, qemu on Arch Linux is a bit different.  I had to elide the line
	-device virtio-gpu-device \
in the riscv64-generic-qemu.sh script

FYI.  Thanks again..