phoenix-rtos/phoenix-rtos-project #861: posix: SIGCHLD signal is set to SIG_IGN zombie shouldn't be generated

agkaminski, 2023-09-27:
From what I see it's not really specified in the POSIX. Something to be studied further

https://stackoverflow.com/questions/18437779/do-i-need-to-do-anything-with-a-sigchld-handler-if-i-am-just-using-wait-to-wai

Not really supported by the linked docs:
[POSIX](http://pubs.opengroup.org/onlinepubs/9699919799/functions/V2_chap02.html#tag_15_04) says about SIG_IGN (strictly under the XSI extension note):

    If the action for the SIGCHLD signal is set to SIG_IGN, child processes of the calling processes shall not be transformed into zombie processes when they terminate. If the calling process subsequently waits for its children, and the process has no unwaited-for children that were transformed into zombie processes, it shall block until all of its children terminate, and wait(), waitid(), and waitpid() shall fail and set errno to [ECHILD].

nalajcie, 2023-09-27:
Ouch, nice finding. The most clear description I see is in `_Exit()` (https://pubs.opengroup.org/onlinepubs/9699919799/functions/_Exit.html):

[[XSI](javascript:open_code('XSI'))] [Option Start] If the parent process of the calling process has set its SA_NOCLDWAIT flag or has set the action for the SIGCHLD signal to SIG_IGN:

* The process' status information (see [Status Information](https://pubs.opengroup.org/onlinepubs/9699919799/functions/V2_chap02.html#tag_15_13)), if any, shall be discarded.

* The lifetime of the calling process shall end immediately. If SA_NOCLDWAIT is set, it is implementation-defined whether a SIGCHLD signal is sent to the parent process.

* If a thread in the parent process of the calling process is blocked in [wait()](https://pubs.opengroup.org/onlinepubs/9699919799/functions/wait.html), [waitpid()](https://pubs.opengroup.org/onlinepubs/9699919799/functions/waitpid.html), or [waitid()](https://pubs.opengroup.org/onlinepubs/9699919799/functions/waitid.html), and the parent process has no remaining child processes in the set of waited-for children, the [wait()](https://pubs.opengroup.org/onlinepubs/9699919799/functions/wait.html), [waitid()](https://pubs.opengroup.org/onlinepubs/9699919799/functions/waitid.html), or [waitpid()](https://pubs.opengroup.org/onlinepubs/9699919799/functions/waitpid.html) function shall fail and set errno to [ECHILD].

Otherwise:

* Status information (see [Status Information](https://pubs.opengroup.org/onlinepubs/9699919799/functions/V2_chap02.html#tag_15_13)) shall be generated.

* The calling process shall be transformed into a zombie process. Its status information shall be made available to the parent process until the process' lifetime ends.

* The process' lifetime shall end once its parent obtains the process' status information via a currently-blocked or future call to [wait()](https://pubs.opengroup.org/onlinepubs/9699919799/functions/wait.html), [waitid()](https://pubs.opengroup.org/onlinepubs/9699919799/functions/waitid.html) (without WNOWAIT), or [waitpid()](https://pubs.opengroup.org/onlinepubs/9699919799/functions/waitpid.html).

* If one or more threads in the parent process of the calling process is blocked in a call to [wait()](https://pubs.opengroup.org/onlinepubs/9699919799/functions/wait.html), [waitid()](https://pubs.opengroup.org/onlinepubs/9699919799/functions/waitid.html), or [waitpid()](https://pubs.opengroup.org/onlinepubs/9699919799/functions/waitpid.html) awaiting termination of the process, one (or, if any are calling [waitid()](https://pubs.opengroup.org/onlinepubs/9699919799/functions/waitid.html) with WNOWAIT, possibly more) of these threads shall obtain the process' status information as specified in [Status Information](https://pubs.opengroup.org/onlinepubs/9699919799/functions/V2_chap02.html#tag_15_13) and become unblocked.

* A SIGCHLD shall be sent to the parent process.

nalajcie, 2023-09-27:
There are also special requirements regarding waking up the `waitpid` threads, not sure we do it according to spec right now:

When status information becomes available to the parent process and more than one thread in the parent process is waiting for the status information (blocked in a call to [wait()](https://pubs.opengroup.org/onlinepubs/9699919799/functions/wait.html), [waitid()](https://pubs.opengroup.org/onlinepubs/9699919799/functions/waitid.html), or [waitpid()](https://pubs.opengroup.org/onlinepubs/9699919799/functions/waitpid.html) with arguments that would match the status information):

* If none of the matching threads is in a call to [waitid()](https://pubs.opengroup.org/onlinepubs/9699919799/functions/waitid.html) with the WNOWAIT flag set in the options argument, the thread that obtains the status information is unspecified.

* Otherwise (at least one of the matching threads is in a call to [waitid()](https://pubs.opengroup.org/onlinepubs/9699919799/functions/waitid.html) with the WNOWAIT flag set), the matching thread or threads that obtain the status information is unspecified except that at least one of the matching threads shall obtain the status information and at most one of the matching threads that are not calling [waitid()](https://pubs.opengroup.org/onlinepubs/9699919799/functions/waitid.html) with the WNOWAIT flag set shall obtain it.

nalajcie, 2023-09-27:
Not related to SIGCHLD, but directly to `posix_died` implementation - SIGHUP generation and sending to children:

* If the process is a controlling process, the SIGHUP signal shall be sent to each process in the foreground process group of the controlling terminal belonging to the calling process.

* If the process is a controlling process, the controlling terminal associated with the session shall be disassociated from the session, allowing it to be acquired by a new controlling process.

* If the exit of the process causes a process group to become orphaned, and if any member of the newly-orphaned process group is stopped, then a SIGHUP signal followed by a SIGCONT signal shall be sent to each process in the newly-orphaned process group.