phoenix-rtos/phoenix-rtos-project #866: armv7m7-imxrt117x-evk: system occasionally freezes while exiting spawned psh

damianloew, 2023-09-28:
- it occurs after executing some of psh tests (for example gibber, auth)
- discovered both on CI and locally
- phrtos-project rev: https://github.com/phoenix-rtos/phoenix-rtos-project/commits/master

[image]

damianloew, 2024-06-10:
On ea760a2fdc7ebb328303851bb29f086b3f4775c0 still occurs, but the behavior is a bit different:

```
(psh)% dmesg -D
(psh)% sysexec psh
(psh)% exit
(psh)% dmesg -E
(psh)% 
(psh)% ps
     PID     PPID PR STATE  %CPU    WAIT        TIME   VMEM THR CMD             
       0        0  4 ready  99.9   8.1ms    00:01:00  71.5K   2 [idle]
       1        0  4 sleep   0.0   705us    00:00:00      0   1 init
       2        1  4 sleep   0.0   519us    00:00:00  18.5K   1 dummyfs
       3        1  2 sleep   0.0   429us    00:00:00  20.5K  10 imxrt-multi
       4        1  4 ready   0.0   142us    00:00:00    31K   1 psh
(psh)% 
(psh)% dmesg -D
(psh)% sysexec psh
(psh)% exit
(psh)% dmesg -E
(psh)% 
(psh)% ps
     PID     PPID PR STATE  %CPU    WAIT        TIME   VMEM THR CMD             
       0        0  4 ready  99.9   8.1ms    00:01:19  71.5K   2 [idle]
       1        0  4 sleep   0.0   705us    00:00:00      0   1 init
       2        1  4 sleep   0.0   519us    00:00:00  18.5K   1 dummyfs
       3        1  2 sleep   0.0   429us    00:00:00  20.5K  10 imxrt-multi
       4        1  4 ready   0.0   142us    00:00:00    31K   1 psh
(psh)% 
(psh)% dmesg -D
```

As it is presented freeze happens while executing one of the next commands

gerard5, 2024-06-10:
@damianloew  Have you tried with the [fix](https://github.com/phoenix-rtos/phoenix-rtos-build/commit/072b8599ebcf4178d0094d3ae256dd4e2eddba0c) in the toolchain build process that was recently merged?

damianloew, 2024-06-10:
> @damianloew Have you tried with the [fix](https://github.com/phoenix-rtos/phoenix-rtos-build/commit/072b8599ebcf4178d0094d3ae256dd4e2eddba0c) in the toolchain build process that was recently merged?

Yeah, it was the reason why it was failing, with a newer toolchain the issue no longer occur, checked on ea760a2fdc7ebb328303851bb29f086b3f4775c0

Probably fixed in https://github.com/phoenix-rtos/phoenix-rtos-build/commit/072b8599ebcf4178d0094d3ae256dd4e2eddba0c (toolchain rebuild needed)

closed 2024-06-10 by damianloew

nalajcie, 2024-06-10:
https://github.com/phoenix-rtos/phoenix-rtos-tests/blob/master/psh/tools/psh.py#L292 should be removed then