phoenix-rtos/phoenix-rtos-project #869: Process keep waiting infinitly when there is no existing unwaited-for child processes.

adamdebek, 2023-09-29:
In below code there are 2 threads waiting for any child to exit, but only 1 child is going to exit, when it does 1 thread should reap a child process and the other one should should stop waiting due to lack of unwaited-for child processes, `wait()` in this process should return `-1` and set `errno` to `ECHILD`. Additionally `pthread_cancel()` couldn't cancel this still waiting thread.
[image]

Code to reproduction (ia32-generic-qemu):

```C
#include <stdio.h>
#include <pthread.h>
#include <signal.h>
#include <unistd.h>
#include <errno.h>
#include <stdlib.h>
#include <string.h>

struct test_ThreadArgs{
	int retWaitThr;
	int errnoThr;
};

static void *test_threadWait(void *args)
{
	int status;
	struct test_ThreadArgs *threadArgs = (struct test_ThreadArgs *)args;

	errno = 0;
	threadArgs->retWaitThr = wait(&status);
	threadArgs->errnoThr = errno;

	/* If all worked one thread should jump in this loop */
	if (threadArgs->retWaitThr == -1) {
		while(1) {
			printf("in thread\n");
		}
	}
	
	return NULL;
}

static void test_sigusrHandler(int signum)
{
	struct sigaction sa;

	sa.sa_handler = SIG_IGN;
	sigaction(SIGUSR1, &sa, NULL);
}

int main(void)
{
	pid_t pid;
	pthread_t tid1, tid2;

	pid = fork();
	/* child */
	if (pid == 0) {
		struct sigaction sa;

		sa.sa_handler = test_sigusrHandler;
		sigemptyset(&sa.sa_mask);
		sa.sa_flags = 0;
		sigaction(SIGUSR1, &sa, NULL);

		pause();
		_exit(EXIT_SUCCESS);
	}
	/* parent */
	else {
		volatile struct test_ThreadArgs argsThr1, argsThr2; 

		memset((void *)&argsThr1, 0, sizeof(struct test_ThreadArgs));
		memset((void *)&argsThr2, 0, sizeof(struct test_ThreadArgs));

		pthread_create(&tid1, NULL, test_threadWait, (void *)&argsThr1);
		pthread_create(&tid2, NULL, test_threadWait, (void *)&argsThr2);

		kill(pid, SIGUSR1);

		while(argsThr1.retWaitThr == 0 && argsThr2.retWaitThr == 0) {
			;
		}

		if(argsThr1.retWaitThr != -1 && argsThr1.errnoThr != ECHILD) {
			//pthread_cancel(tid1);
			fprintf(stderr, "Error: First thread still waiting\n");
		} else if (argsThr2.retWaitThr != -1 && argsThr2.errnoThr != ECHILD) {
			//pthread_cancel(tid2);
			fprintf(stderr, "Error: Second thread still waiting\n");
		}
		else if((argsThr1.retWaitThr != 0 && argsThr1.errnoThr == 0) && (argsThr2.retWaitThr != 0 && argsThr2.errnoThr == 0)) {
			fprintf(stderr, "Error: More than 1 thread unblocked\n");
		}

		/* Assert clean first child */
		kill(pid, SIGKILL);
                /* Loop otherwise main program ends and all threads are destroyed */
		while(1) {
			;
		}
	}

	return 0;
}
```

agkaminski, 2023-09-29:
Behaviour on linux:
`Error: First thread still waiting`
Waiting for any chinld or for the same child from two threads is imho a race condition and a bug.

adamdebek, 2023-09-30:
I updated reproduction code and attached some screenshots. Now we can clearly see that thread keep waiting but should not.