phoenix-rtos/phoenix-rtos-project #871: Atexit not calling function registed inside previously registered function

adamdebek, 2023-10-02:
From Open Group atexit() doc: _At normal program termination, all functions registered by the atexit() function shall be called, in the reverse order of their registration, except that a function is called after any previously registered functions that had already been called at the time it was registered._

Code to reproduction:
```C
#include <stdio.h>
#include <stdlib.h>

void f1(void) {
        printf("1111\n");
}

void f2(void) {
        printf("2222\n");
}

void f3(void) {
        atexit(f1);
        printf("3333\n");
}

int main(void)
{
        atexit(f1);
        atexit(f2);
        atexit(f3);

        return 0;
}
```

ia32-generic-qemu:
[image]

host:
[image]

badochov, 2023-10-11:
@adamdebek I've added support for this, do we have some disabled tests that mey be reenabled?

closed 2023-10-11 by badochov

adamdebek, 2023-10-11:
I will enable it, these tests are still on my pr. Thanks for support addition.