phoenix-rtos/phoenix-rtos-project #888: zynq7000-qemu: errors that happen occasionally in test-exit

damianloew, 2023-10-16:
[image]
[image]

badochov, 2023-10-17:
I've also seen some aborts on zedboard nightly runs, are they also appearing on this test only?

damianloew, 2023-10-17:
> I've also seen some aborts on zedboard nightly runs, are they also appearing on this test only?

Yes, it may be related with the following error that you mentioned (from what I know it occurred only once):
[image]

https://github.com/phoenix-rtos/phoenix-rtos-project/actions/runs/6367034681/job/17285380717

badochov, 2023-10-17:
NOTE: I don't think those errors were present before around the time of signals rework.

adamdebek, 2023-10-18:
This fixes the issue: https://github.com/phoenix-rtos/libphoenix/pull/303

damianloew, 2023-10-18:
Moved to https://github.com/phoenix-rtos/phoenix-rtos-project/issues/891
Will be closed after closing the mentioned issue.

closed 2023-10-18 by adamdebek