phoenix-rtos/phoenix-rtos-project #911: libphoenix: mktemp() wrong divisor value for modulo operation

kemonats, 2023-11-05:
```c
static char pfcs[] = "ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz0123456789._-";
```
https://github.com/phoenix-rtos/libphoenix/blob/8e3bfcd6fa7b4335e666e03a8f0e68bc54cdac99/stdlib/mktemp.c#L27

```c
                tail[i] = pfcs[rand % sizeof(pfcs)];
```
https://github.com/phoenix-rtos/libphoenix/blob/8e3bfcd6fa7b4335e666e03a8f0e68bc54cdac99/stdlib/mktemp.c#L51

Should be sizeof(pfcs) - 1 (without nul character)
In addition, rand should be declared as unsigned.

To improve this function:
```c
	unsigned char rand[6];
	read(fd, &rand, sizeof(rand));
	for (i = 0; i < 6; i++) {
		tail[i] = pfcs[rand[i] % (sizeof(pfcs) - 1)];
	}
```
and then there will be one call to the read() function instead of six

agkaminski, 2023-11-06:
Regarding first remark - IMHO it's ok as is, modulo `sizeof(pfcs)` will give us access to all defined characters - max value is `sizeof(pfcs) - 1`, so the last index.

nalajcie, 2023-11-06:
> Regarding first remark - IMHO it's ok as is, modulo `sizeof(pfcs)` will give us access to all defined characters - max value is `sizeof(pfcs) - 1`, so the last index.

```c
static char  pfcs[] = {'A', 'B', 'C', '\0'}; // the same as = "ABC";
// sizeof(pfcs) == 4

idx = rand % sizeof(pfcs); // assuming 3 
tail[i] = pfcs[idx] = pfcs[3] = '\0'
```
there is no out-of-bounds, but unintended usage of `\0` here

agkaminski, 2023-11-06:
@nalajcie  Ok, now I understand, it's wrong indeed, thx

closed 2023-11-07 by agkaminski