phoenix-rtos/phoenix-rtos-project #928: Readdir() hangs on closed directory

adamdebek, 2023-11-16:
Code to reproduction:
```
#include <dirent.h>
#include <errno.h>
#include <unistd.h>
#include <stdio.h>

int main() {
	mkdir("a", 0700);
        DIR *d = opendir("a");
        closedir(d);

        errno = 0;
        printf("readdir: %p\n", readdir(d));

        printf("errno: %d\n", errno);
}
```

Output host:
[image]
Output imx106 (dummyfs):
[image]
Output zedboard (jffs2):
[image]
Output ia32 (ext2):
Hangs ...
[image]

nalajcie, 2023-11-17:
Specs for closedir says:  _Upon return, the value of dirp may no longer point to an accessible object of the type DIR_ (https://pubs.opengroup.org/onlinepubs/9699919799/functions/closedir.html)

There is no valid operation you can do on `d` after calling `closedir` (you essentially try to access free uninitialized memory) - It's within the specs for your program to abort, hang or perform in unexpected way.

What should not happen is for full OS to freeze or FS to break / became unavailable for other programs - please check if that is the case - if not, please close the issue.

adamdebek, 2023-11-20:
I checked what you requested and it did not result in OS freeze or FS breakdown.

closed 2023-11-20 by adamdebek