phoenix-rtos/phoenix-rtos-project #932: armv7a9-zyq7000-qemu: stucks at various places in tests after launching busybox test suite

damianloew, 2023-11-20:
***DESCRIPTION***

After the following submodules update: c1a93f876635749125ef5fde4a4f41f08af5d3f8 system may stuck while executing tests.
 - The first discovered place is `chdir_tofile` testcase - trying to change directory to file. This one occurs always within CI [sc1]
 
 - When trying to reproduce this issue locally the first ocurrance was in `setvbuf_linebuffer` case - calling setvbuf() in line buffer (_IOLBF) mode [sc2]
 
The next tests may also fail, for example due to stuck just after executing the first test case [sc3]

 - This behavior has been discovered only when running tests in nightly mode, which means that busybox test suite is also launched and there new qemu is spawned before each test.

***REPRODUCTION PATH***

0. Checkout to the following c1a93f876635749125ef5fde4a4f41f08af5d3f8 phrtos-project commit
1. Build the project for `armv7a9-zynq7000-qemu` using:
```
LONG_TEST=y TARGET=armv7a9-zynq7000-qemu phoenix-rtos-build/build.sh clean all test
```
2. Run docker-devel docker image by calling: `./docker-devel.sh`
3. Run the nightly test campaign using the following command to stream output and save additional data to files:
```
python3 phoenix-rtos-tests/runner.py -T armv7a9-zynq7000-qemu --nightly -S -O
```
4. The first fails should be visible when executing libc tests

***SCREENSHOTS***

- sc1
[image]
- sc2
[image]
- sc3
[image]

jmaksymowicz, 2023-11-21:
* The common cause of failures mentioned in the issue seems to be timeout during tests that perform a filesystem operation.
* On my local setup:
    * After fresh build running `libc` test suite on its own results in no errors.
    * When running all tests the first failure is much earlier during `busybox/cp`, testcase 'cp-copies-large-file' due to timeout. After this test failure, subsequent tests timeout waiting for shell.
    * Same as above happens on [phoenix-rtos-project@878d762](https://github.com/phoenix-rtos/phoenix-rtos-project/commit/878d762de5eb57d9758efcabb91c07d757ab7aeb) (before changes mentioned in the issue)
    * During initial tests after building a filesystem image, system initialization (from start until shell prompt appears) takes about 50 seconds, but as more test cases are run on the same image that time increases to about 80 seconds.
    * After increasing timeouts on waiting for reboot and test results, all tests pass. Here is the CSV file with results: [report_longer_timeout.csv](https://github.com/phoenix-rtos/phoenix-rtos-project/files/13428341/report_longer_timeout.csv)

This leads me to the conclusion that the cause of these failures is poor performance of the filesystem. This seems to be a [known, long-standing problem](https://github.com/phoenix-rtos/phoenix-rtos-project/issues/450). It is likely that currently set timeouts are marginal and after adding a new large file to the filesystem (the `zynq7000-sdcard` driver, even though it's unused) timeouts are triggered during tests. Temporarily this could be solved by increasing timeout values. In the long term perhaps the jffs2 driver could be optimized to increase performance.

damianloew, 2023-11-21:
> * The common cause of failures mentioned in the issue seems to be timeout during tests that perform a filesystem operation.
> * On my local setup:
>
>   * After fresh build running `libc` test suite on [...]

Thank you for the investigation! I agree on that, it may be caused indirectly by adding a larger file. However when executing tests on my host, the first error was in `chdir_tofile` case, which took <1s before and now 30s timeout is insufficient, so I am afraid that increasing them won't help :/

damianloew, 2024-10-01:
Not seen anymore - probably fixed, I am closing it for now