phoenix-rtos/phoenix-rtos-project #1004: Possible problems when changing compiler version to >= 10

kemonats, 2024-02-11:
If you plan to change your `gcc` compiler version to `>= 10` then you may experience the following problems:
1. the compiler, in the case of a simple loop writing a constant value to cells of memory may use `memset()` function, which is not defined.
   For example, such loops:

   https://github.com/phoenix-rtos/phoenix-rtos-kernel/blob/1c700bbab4851a4d9010ddc4ff1b14fcada19295/lib/printf.c#L114-L115
   ```c
                while (pad_len-- > 0)
                        *out++ = ' ';
   ```

   https://github.com/phoenix-rtos/phoenix-rtos-kernel/blob/1c700bbab4851a4d9010ddc4ff1b14fcada19295/hal/armv7a/pmap.c#L546-L549
   ```c
        for (i = 0; i < sizeof(pmap_common.asid_map) / sizeof(pmap_common.asid_map[0]); ++i) {
                pmap_common.asid_map[i] = NULL;
                pmap_common.asids[i] = i;
        }
   ```
   Solution:
   - adding the `-ffreestanding` option to the `CFLAGS` in `Makefile.common` or in the `Makefile` in the `kernel` and `plo` repository.

2. since version `10` `gcc`, the default option is `-fno-common` ( `9` has `-fcommon` set as default)
   This causes a problem with linking the `libjffs2` library.    This is caused by the definition of the `init_user_ns` variable as `global` in the header file:
   https://github.com/phoenix-rtos/phoenix-rtos-filesystems/blob/8af595a6b37129b4a6cb7e0cb93705c4d6b3476b/jffs2/phoenix-rtos.h#L139
   ```c
   struct user_namespace init_user_ns;
   ```
   Solution:
   - define the `init_user_ns` variable in one of the source files, and in the header file declare this variable.as `extern`.
   - compile the source files that make up this library with the `-fcommon` option.

nalajcie, 2024-02-11:
Thanks for bringing it to our attention. 

Regarding the (1):
 - IMHO plo / kernel should be compiled with `-ffreestanding` as they are being run in C freestanding environment
 - adding the `-ffreestanding` option is not enough for gcc to avoid `memset` problems (see my comment here: https://github.com/phoenix-rtos/plo/pull/167#issuecomment-1072521885) - it explicitly requires `mem*` functions to be implemented even in freestanding environment

badochov, 2024-09-06:
2) Fixed as of https://github.com/phoenix-rtos/phoenix-rtos-filesystems/pull/125 and https://github.com/phoenix-rtos/phoenix-rtos-build/pull/181