phoenix-rtos/phoenix-rtos-project #1013: armv7m4-stm32l4x6: mkfifo fails to create fifo

mateusz-bloch, 2024-02-13:
`mkfifo` is not creating a named pipe on `stm32`

Code to reproduction:
```C
#include <stdio.h>
#include <stdlib.h>
#include <sys/stat.h>
#include <errno.h>

int main() {
    const char *fifoPath = "/tmp/my_fifo";

    if (mkfifo(fifoPath, S_IRUSR | S_IWUSR) == 0) {
        printf("FIFO '%s' created\n", fifoPath);
    } else {
        printf("mkfifo failed");
    }

    return 0;
}
```

agkaminski, 2024-02-13:
Expected behavior, as named pipes require posixsrv

gerard5, 2024-02-16:
I'm adding libposixsrv, maybe these PRs will solve the issue on stm32l4a6-nucleo evb:

- https://github.com/phoenix-rtos/phoenix-rtos-devices/pull/470
- https://github.com/phoenix-rtos/phoenix-rtos-project/pull/1019
- https://github.com/phoenix-rtos/phoenix-rtos-build/pull/179

mateusz-bloch, 2024-02-20:
Solved

closed 2024-02-20 by mateusz-bloch