phoenix-rtos/phoenix-rtos-project #1017: stm32l4x6: UsageFault in psh test-history

damianloew, 2024-02-15:
[image]

*It doesn't depend on the data that is being sent - this string is randomly generated

- Happens occasionally on CI
- phrtos-project version: 
- to be investigated in detail, the reproduction path has been provided below

### Reproduction path
1. Take a `nucleo-l4a6zg` board and connect it as required in a quickstart
2. build a project for this architecture using the following command:
```
DEBUG=1 TARGET=armv7m4-stm32l4x6-nucleo phoenix-rtos-build/build.sh clean all test
```
3. Flash it to the board using:
```
openocd -f interface/stlink.cfg -f target/stm32l4x.cfg -c "reset_config srst_only srst_nogate connect_assert_srst" -c "program _boot/armv7m4-stm32l4x6-nucleo/phoenix.disk 0x08000000 verify reset exit"
```
4. Edit test history as follows:
[image]
5. Launch the test runner with specifying the proper serial port:
```
python3 runner.py -T armv7m4-stm32l4x6-nucleo -t psh/ -S --no-flash --port /dev/ttyUSB0
```
*You have to click reset button just after launching the command above to provide first prompt for test runner
6. Even after 100-200 attempts you will encounter the error:
[image]

anglov, 2024-02-15:
I would suggest to always put code snippet as text, not as screenshot

damianloew, 2024-02-15:
> I would suggest to always put code snippet as text, not as screenshot
> 
> ```python
>     for i in range(1000):
>         print(f"---ATTEMPT {i}---")
>         assert_multicall_random(p, random_wrapper)
>         assert_clear(p)
> ```

Righjt, thanks for the suggestion

anglov, 2024-02-15:
```
(gdb) bt
#0  0x080108b4 in hal_cpuHalt () at hal/armv7m/arch/cpu.h:162
#1  exceptions_dispatch (n=<optimized out>, ctx=<optimized out>) at /home/anglov/Project/code/ismart-iot/phoenix-rtos-kernel/hal/armv7m/exceptions.c:106
#2  <signal handler called>
#3  _signal_trampoline () at /home/anglov/Project/code/ismart-iot/libphoenix/arch/armv7m/signal.S:23
#4  0x08035414 in psh_runfile (argc=<optimized out>, argv=0x14) at /home/anglov/Project/code/ismart-iot/phoenix-rtos-utils/psh/runfile/runfile.c:36
#5  0x00000000 in ?? ()
```

frame 3
[image]

frame 4
[image]