phoenix-rtos/phoenix-rtos-project #1025: test runner for armv7m4-stm32l4x6-nucleo cannot be run on host env

anglov, 2024-02-23:
I identify some issues not allowing to run tests on my host environment.
1. Target not stopping in plo
```
Flashing an image to device...
Done!
phoenix-rtos-tests/initfini/main: FAIL
PLO ERROR: 
EXPECTED:
(plo)% 
OUTPUT:
,  1900 [ms]Phoenix-RTOS loader v. 1.21 rev: a26067c
hal: Cortex-M4 STM32L4x6
cmd: Executing pre-init script
console: Setting console to 0.1
Waiting for input,     0 [ms]    
Phoenix-RTOS microkernel v. 3.1 rev: 331c7db
hal: STM32 ARMv7 Cortex-M4 r0 p1
hal: softfp, MPU, Thumb
hal: Using NVIC interrupt controller
hal: Using Low-Power Timer
vm: Initializing page allocator 13/320 KB, page_t=12
vm: Initializing memory mapper: (147*80) 11760
vm: Initializing kernel memory allocator: (16*48) 768
vm: Initializing memory objects
proc: Initializing thread scheduler, priorities=8
syscalls: Initializing syscall table [100]
main: Starting syspage programs: 'stm32l4-multi', 'psh'

multidrv: Started
(psh)% 
ADDITIONAL INFO:
OPENOCD OUTPUT:
Open On-Chip Debugger 0.12.0 (2023-03-14-10:37)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
srst_only separate srst_nogate srst_open_drain connect_assert_srst

Info : clock speed 500 kHz
Info : STLINK V3J7M2B4S1 (API v3) VID:PID 0483:374F
Info : Target voltage: 0.025566
Error: target voltage may be too low for reliable debugging
Info : [stm32l4x.cpu] Cortex-M4 r0p1 processor detected
Info : [stm32l4x.cpu] target has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32l4x.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : Unable to match requested speed 500 kHz, using 200 kHz
Info : Unable to match requested speed 500 kHz, using 200 kHz
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
```
2. Alias command after loading app to ram was sometimes corrupted
```
Phoenix-RTOS loader v. 1.21 rev: a26067c
hal: Cortex-M4 STM32L4x6
cmd: Executing pre-init script
console: Setting console to 0.1
Waiting for input,  1500 [ms]    
(plo)% a
phoenix-rtos-tests/ioctl/unit: FAIL

```
3.  Reboot not run (currently required for psh test as openocd for loading app to ram resets target)

Following patch allows me to run test on my PC, but it's definitely not a valid solution. 
```
diff --git a/trunner/harness/plo.py b/trunner/harness/plo.py
index 2b57de3..0fd9d68 100644
--- a/trunner/harness/plo.py
+++ b/trunner/harness/plo.py
@@ -8,6 +8,7 @@ from trunner.text import bold
 from trunner.tools import Phoenixd
 from trunner.types import AppOptions, TestResult, TestStage
 from .base import HarnessError, IntermediateHarness, Rebooter, TerminalHarness
+import time
 
 
 @dataclass(frozen=True)
@@ -77,6 +78,7 @@ class PloInterface:
         except (pexpect.TIMEOUT, pexpect.EOF) as e:
             raise PloError("Failed to go into bootloader mode", output=self.dut.before) from e
 
+        time.sleep(0.5)
         self.dut.send("\n")
 
     def _assert_prompt(self, timeout: Optional[int] = None, error_check=True):
diff --git a/trunner/target/armv7m4.py b/trunner/target/armv7m4.py
index a49b47c..7483a96 100644
--- a/trunner/target/armv7m4.py
+++ b/trunner/target/armv7m4.py
@@ -26,8 +26,15 @@ from .base import TargetBase, find_port
 class ARMv7M4Rebooter(Rebooter):
     # TODO add text mode
     # NOTE: changing boot modes not needed/supported for this target
-    pass
+    def _reboot_dut_text(self, hard):
+        openocd_args = ["-c", "reset_config srst_only srst_nogate connect_assert_srst", "-c", "init;reset;exit"]
 
+        try:
+            with OpenocdGdbServer(interface="stlink", target="stm32l4x", extra_args=openocd_args).run():
+                pass
+        except:
+            pass
+        time.sleep(0.5)
 
 class STM32L4x6OpenocdGdbServerHarness(IntermediateHarness):
     """Harness that runs other harness in Opendocd gdb server context.
@@ -95,6 +102,7 @@ class STM32L4x6PloAppLoader(TerminalHarness, PloInterface):
             path = self.gdb.cwd / Path(app.file)
             sz = path.stat().st_size
 
+            time.sleep(0.5)
             self.alias(app.file, offset=offset, size=sz)
             self.app("ramdev", app.file, "ram", "ram")
```

maska989, 2024-03-15:
Prepared some solutions for point 1 and 3. Not sure about 2.
[Fixed race condition in point where trunner escape PSH](https://github.com/phoenix-rtos/phoenix-rtos-tests/pull/336)
[Added openocd reboot for host environment](https://github.com/phoenix-rtos/phoenix-rtos-tests/pull/337)
Pls, try it and give me feedback.

anglov, 2024-03-19:
In theory it resolves issues 1 and 3 (2 still occurs) but I'am not sure if that's valid solution.

I think dut_text was created for other purposes

https://github.com/phoenix-rtos/phoenix-rtos-tests/blob/909cc0aedad56813f63806e6cbb5bf7389e9d69a/trunner/harness/base.py#L115

I think it's created for rebooting with user interaction.

Also sleep in plo harness is probably workaround. Maybe it's valid to add it at workaround and create issue for analyzing and fixing main problem properly (TBH it's little weird for me why it's failing).

closed 2025-07-28 by anglov