phoenix-rtos/phoenix-rtos-project #1047: Soft reset not occasionally not triggering

adamdebek, 2024-03-29:
On imx6ull occasionally appeared problem with lack of soft restart caused due to too short time of holding `nRST` pin down during test campaigns.

Example run: https://github.com/phoenix-rtos/phoenix-rtos-project/actions/runs/8445210403/job/23132189522

For now previous time (0.1s) is changed to 0.2s, which seemed sufficient but it should be checked more in-depth.

damianloew, 2024-04-17:
To be checked using oscilloscope or firstly by logic analyzer - to see how this signal looks like - maybe rpi started to set this pin more slowly (?)

damianloew, 2024-04-17:
Please underline that it's only needed on imx6ull target - prefix would be a good idea