phoenix-rtos/phoenix-rtos-project #1059: sscanf produces segfault

mateuszkobak, 2024-04-19:
The following code produces a segmentation fault:

```C
char hostname[60] = "[2001:aaa:aaaa:a::a]:8000";
char endbracket;
int len;

sscanf(hostname, "[%*45[0123456789abcdefABCDEF:.]%c%n", &endbracket, &len);
printf("%s\n%c\n%d\n", hostname, endbracket, len);
```

This is a reproduction of a situation in function `Curl_parse_port` in `curl-7.64.1/lib/urlapi.c`.

anglov, 2024-04-19:
1. Please create issues only in phoenix-rtos-project
2. At the first glance reproduction code is valid (working as expected on linux glibc environment under --std=c99 and -pedantic), so probably there is some issue

Side note: curl upstream majorly modified Curl_parse_port function

nalajcie, 2024-04-19:
> ```
> char hostname[60] = "[2001:aaa:aaaa:a::a]:8000";
> int len;
> 
> sscanf(hostname, "[%45[0123456789abcdefABCDEF:.]]%n", hostname, &len);
> printf("%s\n%d\n", hostname, len);
> ```
> 
> Howcome you need the "endbracket" variable? The code above might help

Please note that this is a reproduction code from `curl`, so the intent is not to change the code but check why it fails for our implementation of `sscanf`.

Regarding why the `]` is not used directly - see the original code (https://github.com/curl/curl/blob/curl-7_64_1/lib/urlapi.c#L501) - it's used to detect the zoneidx encoding (eg. `[fe80::20c:29ff:fe9c:409b%eth0]:1234` (if endbracket == '%')

closed 2024-11-06 by Darchiv