phoenix-rtos/phoenix-rtos-project #1073: vm: `_page_map()` should be static

niewim19, 2024-05-10:
`phoenix-rtos-kernel/vm/page.c:_page_map()` should be static as it is not used outside of its file and is not in any header file.

Function: https://github.com/phoenix-rtos/phoenix-rtos-kernel/blob/f29031f5ca6707f70e2720819f1255828a0c5cba/vm/page.c#L329C5-L329C14

astalke, 2024-05-15:
I think `_page_initSizes` should be static as well

niewim19, 2024-05-15:
> I think `_page_initSizes` should be static as well

 - https://github.com/phoenix-rtos/phoenix-rtos-project/issues/1071 yeah, there`s already an issue on that :)

astalke, 2024-05-15:
> > I think `_page_initSizes` should be static as well
> 
>     * [vm: `_page_initSizes()` should be static #1071](https://github.com/phoenix-rtos/phoenix-rtos-project/issues/1071) yeah, there`s already an issue on that :)

Sorry, I have no idea how I could have missed it.

closed 2024-05-24 by Darchiv