phoenix-rtos/phoenix-rtos-project #1078: kernel: crash when a signal is sent to a child process immediately after fork()

jmaksymowicz, 2024-05-16:
Sending a signal such as SIGINT to a process executing `sys_fork()` (or soon after) causes the kernel to crash.
To test this I run `test_busybox which` and try to interrupt it soon after starting the test.

Result on ia32-generic-qemu:
[image]

Result on armv7a9-zynq7000-qemu:
[image]

Both crashes happen during execution of `hal_memcpy` because data is copied into some invalid pointer. 

I managed to track down the bug to the fact that in function `syscalls_dispatch` after executing `proc_fork()` the child process has garbage data in its kernel stack while the function `threads_setupUserReturn` expects a valid userspace CPU context to be at the top of the kernel stack. 

In the function [`proc_fork`](https://github.com/phoenix-rtos/phoenix-rtos-kernel/blob/509945e05ee1da4c22bde4ec1833007c17711306/proc/process.c#L1532) a new kernel stack is set for the child process, but the contents of the parent stack are not copied and it contains unexpected data. 

On a side note: Only the signal checking code uses this incorrectly initialized kernel stack, which is why the code functions correctly if there is no need to handle signals. The stack pointer register in CPU is still pointing to the parent stack, so without any signals the child process exits syscall and restores context correctly.

agkaminski, 2024-05-16:
Solved in https://github.com/phoenix-rtos/phoenix-rtos-kernel/pull/549

jmaksymowicz, 2024-05-16:
Argh, I started investigating the bug just before the fix was merged... At least I learned about how process forking works :)

closed 2024-05-16 by jmaksymowicz