phoenix-rtos/phoenix-rtos-project #1082: vm: `vm_pageFreeAt()` is unreachable code

niewim19, 2024-05-20:
`vm_pageFreeAt()` is a function present in `phoenix-rtos-kernel/vm/page.h` (and implemented in adjacent `page.c` and `page-nommu.c`), yet it is seems to be executed nowhere, which makes it an unreachable code (as it`s probably removed by a compiler). According to MISRA rule 2.1 it shall be removed.

Here: https://github.com/phoenix-rtos/phoenix-rtos-kernel/blob/f29031f5ca6707f70e2720819f1255828a0c5cba/vm/page.c#L189

astalke, 2024-05-21:
I've checked my compiled binary on `ia32-generic-qemu` with `objdump` and there is no `vm_pageFreeAt`, so it must have been removed by the compiler.

agkaminski, 2024-05-21:
@astalke so it's a dead code and shall be removed. Perhaps this could be done as a part of https://github.com/phoenix-rtos/phoenix-rtos-kernel/pull/550?

astalke, 2024-05-21:
> @astalke so it's a dead code and shall be removed. Perhaps this could be done as a part of [phoenix-rtos/phoenix-rtos-kernel#550](https://github.com/phoenix-rtos/phoenix-rtos-kernel/pull/550)?

I'll include that in that PR as a new commit. It is also the only reasonable use of `_page_free`, so I think I can merge it with `page_free`. I think it also fixes https://github.com/phoenix-rtos/phoenix-rtos-project/issues/1080

closed 2024-05-24 by Darchiv