phoenix-rtos/phoenix-rtos-project #1084: Process sleeping on accept() can't be killed

adamdebek, 2024-05-21:
Reproduction:
Run this hello program
```C
#include <stdio.h>
#include <string.h>
#include <unistd.h>
#include <stdlib.h>
#include <stdint.h>
#include <sys/wait.h>
#include <sys/socket.h>
#include <arpa/inet.h>
#include <netinet/in.h>

int main(void)
{
	pid_t pid;
	int sockfd;
	struct sockaddr_in dest_addr;
	uint16_t dest_port = 2000;

	pid = fork();
	if (pid < 0) {
		perror("fork");
		exit(EXIT_FAILURE);
	} else if (pid == 0) {
		int con;

		if ((sockfd = socket(AF_INET, SOCK_STREAM, IPPROTO_TCP)) < 0) {
			perror("socket");
			close(sockfd);
			exit(EXIT_FAILURE);
		}

		dest_addr.sin_family = AF_INET;
		dest_addr.sin_addr.s_addr = inet_addr("127.0.0.1");
		dest_addr.sin_port = htons(dest_port);

		if (bind(sockfd, (struct sockaddr*)&dest_addr, sizeof dest_addr) < 0) {
			perror("bind");
			close(sockfd);
			exit(EXIT_FAILURE);
		}

		if (listen(sockfd, 100) < 0 ) {
			perror("listen");
			close(sockfd);
			exit(EXIT_FAILURE);
		}

		socklen_t len = sizeof dest_addr;
		if ((con = accept(sockfd, (struct sockaddr *)&dest_addr, &len)) < 0) {
			perror("accept");
			close(sockfd);
			close(con);
			exit(EXIT_FAILURE);
		}

		close(sockfd);
		close(con);
	} else {
		exit(EXIT_SUCCESS);
	}

	return 0;
}
```
Then try to kill it using `kill -9 <pid>`
My output (ia32):
[image]

nalajcie, 2024-05-21:
All blocking IO operations are uninterruptible right now (like `read()`, `write`, `select`, `accept`, etc.). There are no immediate plans to fix it, as it's a very complex matter to solve.

The workaround for user code is to either use non-blocking IO or use `poll`/`select` with timeout.

adamdebek, 2024-05-21:
Ok, I just noticed it by chance and thought it is unintended.