phoenix-rtos/phoenix-rtos-project #1086: zynq-zedboard: exception abort loop during system startup

damianloew, 2024-05-27:
- Happens occasionally
 - Issue firstly discovered on 0a9dfbc (to be investigated whether it can be reproduced using earlier commits)
 - For now reproduced only on `u1 zedboard` test unit on CI: https://github.com/phoenix-rtos/phoenix-rtos-project/actions/runs/9249488896/job/25441474802

- Output:
```
dummyfs: initialized
version 2.2. (NAND) (SUMMARY)  © 2001-2006 Red Hat, Inc.

Exception: 4 #Abort
 r0=00000000  r1=00000000  r2=000000b0  r3=00027ab8
 r4=00026730  r5=00000000  r6=00000003  r7=7fffffa8
 r8=00000000  r9=c0149ab4 r10=c0149ab8  fp=c0161f8c
 ip=fffffffc  sp=c0161ef4  lr=00019b55  pc=00015702
psr=20000030 dfs=00000807 dfa=00000000 ifs=00000000
ifa=00000000

Exception: 4 #Abort

Exception: 4 r0=00000000  r1=00000000  r2=000000b0  r3=00027ab8
 r4=00026730  r5=00000000  r6=00000003  r7=7fffffa8
 r8=00000000  r9=c0149ab4 r10=c0149ab8  fp=c0161f8c
 ip=fffffffc  sp=c0161ef4  lr=00019b55  pc=00015702
psr=20000030 dfs=00000807 dfa=00000000 ifs=00000000
ifa=00000000

 #Abort
 r0=00000000  r1=00000000  r2=000in thread 18, process "/bin/bind" (PID: 8)
000b0  r3=00027ab8
 r4=00026730  r5=00000000  r6=00000003  r7=7fffffa8
 r8=00000000  r9=c0149ab4 r10=c0149ab8  fp=c0161f8c
 ip=fffffffc  sp=c0161ef4  lr=00019b55  pc=00015702
psr=20000030 dfs=00000807 dfa=00000000 ifs=00000000
ifa=00000000
```

anglov, 2024-05-29:
arm-phoenix-addr2line 0x00019b55 -e _build/armv7a9-zynq7000-zedboard/prog/psh 
/src/libphoenix/stdlib/atexit.c:51
arm-phoenix-addr2line 0x00015702 -e _build/armv7a9-zynq7000-zedboard/prog/psh 
/src/libphoenix/arch/arm/memset.S:131

agkaminski, 2024-05-29:
> arm-phoenix-addr2line 0x00019b55 -e _build/armv7a9-zynq7000-zedboard/prog/psh /src/libphoenix/stdlib/atexit.c:51 arm-phoenix-addr2line 0x00015702 -e _build/armv7a9-zynq7000-zedboard/prog/psh /src/libphoenix/arch/arm/memset.S:131

I've came to same solution by compiling the artifacts myself, but it seems to me inconsistent with the character of the exception (instruction that `pc` is pointing to is not accessing memory, but the abort was caused by NULL dereference). I suspect there was some difference between binary on CI and one I compiled myself.

anglov, 2024-05-29:
> > arm-phoenix-addr2line 0x00019b55 -e _build/armv7a9-zynq7000-zedboard/prog/psh /src/libphoenix/stdlib/atexit.c:51 arm-phoenix-addr2line 0x00015702 -e _build/armv7a9-zynq7000-zedboard/prog/psh /src/libphoenix/arch/arm/memset.S:131
> 
> I've came to same solution by compiling the artifacts myself, but it seems to me to be inconsistent with the character of the exception (instruction that `pc` is pointing to is not accessing memory, but the abort was caused by NULL dereference). I suspect there was some difference between binary on CI and the one I compiled myself.

I compare md5sum of stripped binary and it's same

damianloew, 2024-11-05:
[testfile.zip](https://github.com/user-attachments/files/17635030/testfile.zip)