phoenix-rtos/phoenix-rtos-project #1091: tests: wrong message in case of failing assert in harness psh tests

damianloew, 2024-06-04:
Reproduction:
1. Modify test-prompt as follows:
```
import psh.tools.psh as psh

@psh.run
def harness(p):
    assert False
    # psh.init() already asserts prompt
    pass
```
2. Run the test, it can be `ia32-generic-qemu`
```
./phoenix-rtos-tests/runner.py --target ia32-generic-qemu -t phoenix-rtos-tests/psh/
```
3. The result will be:
```
flash: OK
phoenix-rtos-tests/psh/prompt: FAIL
SHELL ERROR: Couldn't find a prompt!
EXPECTED: (psh)% 
OUTPUT:
TESTS: 2 PASSED: 1 FAILED: 1 SKIPPED: 0
```

nalajcie, 2024-06-07:
IMHO we shouldn't send extra newline, each test is a program and should finish as such. Adding extra newline doesn't solve the underlying problem (and looks ugly/unintentional in test logs).

I think the problem is with `psh.run` wrapper which doesn't deinitialize DUT correctly on exception (eg. it never exits the spawned psh).

mateusz-bloch, 2024-06-07:
Right, I see that deinit is not always executed. This will be changed in the above PR.

closed 2024-06-10 by damianloew