phoenix-rtos/phoenix-rtos-project #1095: Intermittent MemMgtFault on stm32l4x6-nucleo

agkaminski, 2024-06-14:
For now seen once on CI (https://github.com/phoenix-rtos/phoenix-rtos-project/actions/runs/9502217223/job/26189746156):
```
 phoenix-rtos-tests/psh/history: FAIL
SHELL ERROR: Couldn't find a prompt!
EXPECTED: (psh)% 
OUTPUT:
ciE3fGUFgW2uIV~_

Exception: 4 #MemMgtFault
 r0=00000001  r1=00000071  r2=00000071  r3=40004400
 r4=2000d434  r5=2000d150  r6=2000d370  r7=2000d430
 r8=00000002  r9=20009d30 r10=2000d32c r11=20013e88
r12=00000001 psr=21000000  lr=0802195d  pc=08021966
psp=2000d318 msp=20017200 exr=fffffffd bfa=40004400
cfs=00000082 mma=40004400
```

Seems like psh might be trying to access USART2 CR1 register for some reason. Did not look further into this issue.

damianloew, 2024-06-17:
- Discovered for a second time on 5eabbc3933641af6b1ff35fc3da6095455023a73 with the following output:

```
Exception: 4 #MemMgtFault  
r0=00000001  r1=0000003f  r2=0000003f  r3=40004400
r4=2000d434  r5=2000d150  r6=2000d370  r7=2000d430
r8=00000002  r9=20009d30 r10=2000d32c r11=20013e88
r12=00000001 psr=21000000  lr=0802195d  pc=08021966
psp=2000d318 msp=20017200 exr=fffffffd bfa=40004400
cfs=00000082 mma=40004400
```
https://github.com/phoenix-rtos/plo/actions/runs/9546062432/job/26308199690

 - Seems to happen from time to time

agkaminski, 2024-06-17:
There's a slight chance that https://github.com/phoenix-rtos/phoenix-rtos-kernel/pull/561 might fix it

anglov, 2024-06-17:
PC:
```c
269     static int tty_uartenabled(tty_ctx_t *ctx)
270     {
271             return (*(ctx->base + cr1) & 1) ? 1 : 0;
   0x08021962 <+530>:   ldr.w   r3, [r5, #512]  @ 0x200
   0x08021966 <+534>:   ldr     r3, [r3, #0]

285                             while (((tty_dmarxready(ctx) == 0) && (tty_dmatxready(ctx) == 0)) || (tty_uartenabled(ctx) == 0)) {
   0x08021968 <+536>:   lsls    r3, r3, #31
   0x0802196a <+538>:   bpl.w   0x80217c2 <tty_dmathread+114>
   0x0802196e <+542>:   b.n     0x8021804 <tty_dmathread+180>

235                     libtty_wake_reader(&ctx->ttyCommon);
   0x08021970 <+544>:   mov     r0, r6
   0x08021972 <+546>:   bl      0x8024048 <libtty_wake_reader>
   0x08021976 <+550>:   b.n     0x80218f0 <tty_dmathread+416>
   0x08021978 <+552>:   lsls    r0, r2, #3
   0x0802197a <+554>:   movs    r0, r0
```
LR:

```c
257     static int tty_dmatxready(tty_ctx_t *ctx)
258     {
259             return ((ctx->data.dma.txDoneFlag != 0) && (libtty_txready(&ctx->ttyCommon) != 0)) ? 1 : 0;
   0x08021956 <+518>:   mov     r0, r6
   0x08021958 <+520>:   bl      0x8023d5c <libtty_txready>
   0x0802195c <+524>:   cmp     r0, #0
   0x0802195e <+526>:   beq.w   0x80217c2 <tty_dmathread+114>

260     }
```

```
(gdb) info registers 
r0             0x1                 1
r1             0x39                57
r2             0x39                57
r3             0x40004400          1073759232
r4             0x2000d434          536925236
r5             0x2000d150          536924496
r6             0x2000d370          536925040
r7             0x2000d430          536925232
r8             0x1                 1
r9             0x20009d30          536911152
r10            0x2000d32c          536924972
r11            0x20013e88          536952456
r12            0x1                 1
sp             0x2000d318          0x2000d318
lr             0x802195d           134355293
pc             0x8021966           0x8021966 <tty_dmathread+534>
xPSR           0x21000000          553648128
fpscr          0x0                 0
msp            0x20017200          0x20017200
psp            0x2000d318          0x2000d318
primask        0x0                 0
basepri        0x0                 0
faultmask      0x0                 0
control        0x3                 3
```

Looks like code path that are reached in normal work and works fine.

adamdebek, 2024-06-18:
I checked on `5beec45` and errors still occur. In my opinion it looks like error pertaining to process creation.

[image]

Darchiv, 2024-06-18:
> I checked on `5beec45` and errors still occur. In my opinion it looks like error pertaining to process creation. [...]

Maybe you can try https://github.com/phoenix-rtos/phoenix-rtos-kernel/pull/553?

adamdebek, 2024-06-18:
> > I checked on `5beec45` and errors still occur. In my opinion it looks like error pertaining to process creation.
> > ![image](https://private-user-images.githubusercontent.com/108683715/340650506-4e7199b6-2584-43d0-b7b5-894fbeda86c1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg3MDk1NzcsIm5iZiI6MTcxODcwOTI3NywicGF0aCI6Ii8xMDg2ODM3MTUvMzQwNjUwNTA2LTRlNzE5OWI2LTI1ODQtNDNkMC1iN2I1LTg5NGZiZWRhODZjMS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjE4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYxOFQxMTE0MzdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yMzQyYjUxOWE5ZTNjYjdiNmY5ZDYzYmUzODI4MGI0YjNlNmY4NmFkOGZmMDMwYzI2NDI3ODE3M2FhZmE1MDcxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.jedU6NC6eqa_DjiiNMl1KWFXB-2cPgW_KPBmxKjiN4A)
> 
> Maybe you can try [phoenix-rtos/phoenix-rtos-kernel#553](https://github.com/phoenix-rtos/phoenix-rtos-kernel/pull/553)?

I see it failed during merge to master, so I assume this pr needs further changes.

anglov, 2024-06-19:
At the first glance it looks fixed by https://github.com/phoenix-rtos/phoenix-rtos-kernel/pull/562

@adamdebek could you also check?

adamdebek, 2024-06-19:
> At the first glance it looks fixed by [phoenix-rtos/phoenix-rtos-kernel#562](https://github.com/phoenix-rtos/phoenix-rtos-kernel/pull/562)
> 
> @adamdebek could you also check?

Yea, tests are running already.

adamdebek, 2024-06-20:
I ran over 200 test companies during night and not single one failed, so this confirm the issue is resolved.

closed 2024-06-20 by adamdebek