phoenix-rtos/phoenix-rtos-project #1103: `armv7m7-imxrt117x`: system freeze after using `sysexec psh`, soon after system startup

mateusz-bloch, 2024-06-21:
The system freezes after entering 'sysexec psh' as soon after system startup. This issue arises with following the change [armv7m/pmap: Add missing spinlock](https://github.com/phoenix-rtos/phoenix-rtos-kernel/commit/1508c4384a21f7fbf938031d8f1940a26d8bd0ee) in the kernel.

Example:
```
�Phoenix-RTOS loader v. 1.21 rev: 8e760cc
hal: Cortex-M i.MX RT117x
dev/usb: Initializing usb-cdc(1.2)
dev/flash: Initialized flexspi1
dev/flash/nor: Probing flash id 0x0018709d on port 1.0
dev/flash/nor: Configured ISSI IS25WP128 16MB nor flash(2.0)
dev/pipe: Initialized rtt channel(8.0)
cmd: Executing pre-init script
console: Setting console to 0.0
Waiting for input,     0 [ms]    
Phoenix-RTOS microkernel v. 3.2 rev: 1508c43
hal: NXP i.MX RT117x ARMv7 Cortex-M7 r1 p2
hal: FPU, MPU, Thumb
hal: Using NVIC interrupt controller
hal: Using General Purpose Timer
vm: Initializing page allocator 10/160 KB, page_t=12
vm: Initializing memory mapper: (72*80) 5760
vm: Initializing kernel memory allocator: (16*48) 768
vm: Initializing memory objects
proc: Initializing thread scheduler, priorities=8
syscalls: Initializing syscall table [100]
main: Starting syspage programs: 'dummyfs', 'imxrt-multi', 'psh'
dummyfs: initialized
(psh)% sysexec psh

```

agkaminski, 2024-06-21:
Only thing that comes to my mind regarding this issue is that perhaps syspage mpu data is overwritten after init. Could you please check this change?
https://github.com/phoenix-rtos/phoenix-rtos-kernel/pull/563

mateusz-bloch, 2024-06-21:
> @mateusz-bloch Only thing that comes to my mind regarding this issue is that perhaps syspage mpu data is overwritten after init. Could you please check this change? [phoenix-rtos/phoenix-rtos-kernel#563](https://github.com/phoenix-rtos/phoenix-rtos-kernel/pull/563)

After a quick look, I don't see any changes regarding this issue.