phoenix-rtos/phoenix-rtos-project #1133: reading STDIN freezes a system

damianloew, 2024-08-21:
- Reproduction path:

```
#include <stdio.h>

int main() {
    char c;
    c = getchar();

    return 0;
}
```

 - applies to the reast of functions like scanf(), fgets() that reads from stdin and so on

damianloew, 2024-08-21:
The issue happens after applying the following commit:
[ea15 (HEAD) unistd/file: provide safe write/read/open/close functions](https://github.com/phoenix-rtos/libphoenix/commit/ea158dbf0092d0c9e79baf53e43ec012b0a1d96e)

lukileczo, 2024-08-21:
Seems to be fixed by @ziemleszcz in https://github.com/phoenix-rtos/libphoenix/pull/376

badochov, 2024-09-06:
Shall we close it?

nalajcie, 2024-09-06:
The mentioned PR is not yet merged and the issue still persists on master so please keep it open.

The PR doesn't yet fix all of the issues, so it's not merge-ready.