phoenix-rtos/phoenix-rtos-project #1171: armv8r52-mps3an536-qemu little kernel memory map

adamdebek, 2024-09-20:
around 3GB of ddr memory is declared:
[image]
But only 431KB is available:
[image]

mateusz-bloch, 2024-09-27:
Behavior on the current master is as follows when trying to run` armv8r52-mps3an536-qemu.sh`

```
Phoenix-RTOS loader v. 1.21 rev: 296a9d4
hal: Cortex-R52 MPS3-AN536
cmd: Executing pre-init script
console: Setting console to 0.2
Cannot allocate memory for 'phoenix-armv8r52-mps3an536.elf'
```

agkaminski, 2024-10-01:
Seems to be working fine as of https://github.com/phoenix-rtos/phoenix-rtos-project/commit/bacc9cda61fb6d4be663c7fa781c5cbb7d108ce5

Original issue probably fixed in https://github.com/phoenix-rtos/phoenix-rtos-kernel/commit/74b869b07f93a8fc4dc8d81093f85a34677ecc6b

So I'm closing the issue

closed 2024-10-01 by agkaminski