phoenix-rtos/phoenix-rtos-project #1194: imxrt117x: MemMgtFault on exit command

maska989, 2024-10-04:
One of our previous target's on runner CI was `pimxrt1176` and this was experimental evk. After some time, we needed to replace it with normal `imxrt1176`, because it has some problems not related to normal revision. After adjust runner to it, I found there is problem when `exit` command is executed.

Problem in runner:

[image]

After further investigation:

[image]

As we can see, runner don't report problem because it `exit` close `psh` but not in intentional way.

damianloew, 2024-10-04:
I think to better show what is going on here we can change the name to `armv7m7-imxrt117x: MemMgtFault on exit command from newly spawned psh`

agkaminski, 2024-10-04:
@maska989  This is a litmus paper of old toolchain :) On the current toolchain version this should not occur

damianloew, 2024-10-04:
> @maska989 This is a litmus paper of the old toolchain :) On the current toolchain version this should not occur

It seems to be true, thanks for the quick reaction. Let's wait for @maska989 confirmation

mateusz-bloch, 2024-10-04:
Everything works. Tested on the toolchain from the newest docker build image.

closed 2024-10-04 by agkaminski