phoenix-rtos/phoenix-rtos-project #1225: `/bin/bind` armv7a9-zynq7000-zedboard nightly fail

adamdebek, 2024-11-05:
During nightly on armv7a9-zynq7000-zedboard on `3cef97a` project hash, one of psh tests failed with error:

Exception: 4 #Abort
r0=00024008  r1=00000030  r2=00000ff8  r3=00000048
r4=00024008  r5=00000030  r6=00000031  r7=00000030
r8=00024000  r9=c019df8c r10=c0198678  fp=c019867c
ip=00000001  sp=7fffff20  lr=0001428b  pc=000140b8
psr=20000030 dfs=00000007 dfa=00000040 ifs=0000140f
ifa=ffffffff

in thread 19, process "/bin/bind" (PID: 8)

addr2line output:
arm-phoenix-addr2line 0x000140b8 -e _build/armv7a9-zynq7000-zedboard/prog/psh  
/src/libphoenix/stdlib/malloc_dl.c:238
arm-phoenix-addr2line 0x0001428b -e _build/armv7a9-zynq7000-zedboard/prog/psh  
/src/libphoenix/stdlib/malloc_dl.c:78

To reproduce first build: `DEBUG=1 TARGET=armv7a9-zynq7000-zedboard ./docker-build.sh clean all`
and then run psh tests using trunner many times (even few hundreds)

adamdebek, 2024-11-05:
**My notes**: When I built project without docker image I bumped into same error, but with diffrent program counter and warning from jffs2

warning: (7) jffs2_sum_scan_sumnode: Summary node crc error, skipping summary information.

Exception: 4 #Abort
 r0=00000000  r1=000257e4  r2=00000000  r3=00000000
 r4=7fffffb4  r5=00000000  r6=00000003  r7=00000000
 r8=00000000  r9=00000000 r10=c019c8f8  fp=c019c8fc
 ip=00000000  sp=7fffff80  lr=000167ff  pc=00016802
psr=60000030 dfs=00000807 dfa=00000000 ifs=0000140f
ifa=ffffffff

in thread 19, process "/bin/bind" (PID: 8)

addr2line output:
~/toolchains/arm-phoenix/bin/arm-phoenix-addr2line 0x00016802 -e _build/armv7a9-zynq7000-zedboard/prog/psh 
/home/adam/work6/phoenix-rtos-project/libphoenix/misc/init.c:33

~/toolchains/arm-phoenix/bin/arm-phoenix-addr2line 0x000167ff -e _build/armv7a9-zynq7000-zedboard/prog/psh 
/home/adam/work6/phoenix-rtos-project/libphoenix/misc/init.c:32

In my opinion it looks like fail related to file system not fully mounted and starting executable from it, causing corrupted process image to be run. (/bin/bind is first process run in rc.psh)

damianloew, 2024-11-06:
Please note that `LONG_TEST` env variable is also set during nightly campaign