phoenix-rtos/phoenix-rtos-project #1245: Only regular files timestamps `[amc]time` are updated

adamdebek, 2024-12-03:
From [`fstat()`](https://pubs.opengroup.org/onlinepubs/009696599/functions/fstat.html) open group description:
_The structure members st_mode, st_ino, st_dev, st_uid, st_gid, st_atime, st_ctime, and st_mtime shall have meaningful values for all other file types defined in this volume of IEEE Std 1003.1-2001._

nalajcie, 2024-12-03:
IMHO meaningful doesn't necessarily mean "the same as for regular files", eg on linux: 
[image]

(the same for `/dev/zero`, `/dev/random`; I see that `/dev/ptmx` does actually update [cm]time on creating new PTY).

If you have some specific special files which indeed should trigger `times` update on your mind - please share them :)

adamdebek, 2024-12-03:
I particulary think about fifo, anonymous pipe, sockets and tty's. I see in [here](https://github.com/phoenix-rtos/phoenix-rtos-kernel/blob/master/posix/posix.c#L1188) that message is only sent in case of regular files.

I needed to comment out one of test purpose for anonymous pipes:
[image]

rmikielis, 2026-10-09:
Updating [am]time for servers w/o built-in filesystem (most of devices, posixsrv) would require sending additional message (mtSetAttr) to filesystem server which hosts given file, with request to update required time. It would require additional msg for every ```posix_read```, ```posix_write```, what for RTOS seems unreasonably high additional latency for I/O rw operations.
We consider it OS limitation with current architecture.

closed 2026-11-12 by damianloew