phoenix-rtos/phoenix-rtos-project #1256: dummyfs `readdir()` return every second entry

adamdebek, 2025-01-10:
Reproduction (armv7m7-imxrt106x-evk):
```C
#include <errno.h>
#include <stdlib.h>
#include <unistd.h>
#include <stdio.h>
#include <dirent.h>
#include <string.h>
#include <limits.h>

int main(void)
{
	char entryPath[PATH_MAX];
	DIR *dir;
	struct dirent *entry;
    char *dirPath = "/test";

	if ((dir = opendir(dirPath)) == NULL) {
		perror("opendir");
		return -1;
	}

	while ((entry = readdir(dir)) != NULL) {
        if (!strcmp(".", entry->d_name) || !strcmp("..", entry->d_name)) {
            continue;
        }

        sprintf(entryPath, "%s/%s", dirPath, entry->d_name);

        printf("%s\n", entryPath);

        if (unlink(entryPath) < 0) {
            perror("unlink");
            closedir(dir);
            return -1;
        }
    }

	if (closedir(dir) < 0) {
		perror("closedir");
		return -1;
	}

	/* Remove now empty directory */
	if (rmdir(dirPath) < 0) {
		perror("rmdir");
		return -1;
	}

	return 0;
}
```

[image]

agkaminski, 2025-01-11:
IMHO wrong usage - unlink causes readdir iterator to be de facto invalidated, it's expected for you to read every second file only - you remove id 2, id 3 becomes 2, 4 becomes 3, so when you read id 3 you get a file that was id 4, skipping id 3.

https://stackoverflow.com/questions/1676522/delete-files-while-reading-directory-with-readdir

agkaminski, 2025-01-11:
BTW, `rm *` works in a different way - first `*` is unwind to a list of all files and this list is passed to the `rm`. It does not remove files while reading dir content.

adamdebek, 2025-01-13:
I think this problem is mentioned here: https://github.com/phoenix-rtos/phoenix-rtos-project/issues/900.

Deleteing one of the entries should not affect return of any other entries.

closed 2025-01-13 by adamdebek