phoenix-rtos/phoenix-rtos-project #1258: libphoenix: NAME_MAX is not checked properly

badochov, 2025-01-20:
NAME_MAX is not checked during open when file is not create with parameters `O_WRONLY || O_RDWR` (only then `stat` checking the length is executed). Following code result in no error on zynq, in error due to being longer than ext2's max name len and with a complete crash on armv8r. 

Moreover grepping over `PATH_MAX`  in libphoenix shows that it is seldom checked, so I believe there are more issue simmilar to this one.

```
#include <stdio.h>
#include <fcntl.h>
#include <unistd.h>
#include <limits.h>
#include <string.h>
#include <stdlib.h>

int main(void)
{
	char *name = malloc(NAME_MAX * 2 + 1);
	if (name == NULL) {
		return 2;
	}

	memset(name, NAME_MAX * 2, 'a');
	name[0] = '/';
	name[NAME_MAX * 2] = '\0';

	int res = open(name, O_RDONLY | O_CREAT | O_TRUNC, 0644);

	return 0;
}
```

adamdebek, 2025-10-22:
More functions with same issue.

Reproduction (ia32-generic-qemu):
```c
#include <stdio.h>
#include <string.h>
#include <stdlib.h>
#include <errno.h>
#include <unistd.h>
#include <sys/stat.h>
#include <fcntl.h>

int main(void)
{
	FILE *f;
	int ret;
	long len;
	char *name;
	struct stat s;

	if ((len = pathconf(".", _PC_NAME_MAX)) < 0 && errno != 0) {
		perror("pathconf");
		exit(EXIT_FAILURE);
	}

	// exceed name by 1 byte
	len++;

	if ((name = malloc(len + 1)) == NULL) {
		perror("malloc");
		exit(EXIT_FAILURE);
	}

	memset(name, 'a', len);
	name[len] = '\0';

	errno = 0;
	f = fopen(name, "w");
	printf("fopen ret: %p errno: %d\n", f, errno);

	errno = 0;
	ret = stat(name, &s);
	printf("stat ret: %d errno: %d\n", ret, errno);

	errno = 0;
	ret = remove(name);
	printf("remove ret: %d errno: %d\n", ret, errno);

	errno = 0;
	ret = rmdir(name);
	printf("rmdir ret: %d errno: %d\n", ret, errno);

	errno = 0;
	ret = unlink(name);
	printf("unlink ret: %d errno: %d\n", ret, errno);

	errno = 0;
	ret = link(name, "link");
	printf("link ret: %d errno: %d\n", ret, errno);

	errno = 0;
	ret = rename(name, "new_name");
        printf("rename ret: %d errno: %d\n", ret, errno);

	errno = 0;
	ret = chmod(name, 0664);
	printf("chmod ret: %d errno: %d\n", ret, errno);

	return 0;
}
```

[image]

should be:

[image]

rmikielis, 2026-09-11:
As for today, version on master returns errno 36 on fopen with too long name.
However there is no feasible way of fixing other methods errnos with current architecture.
If file was not created, all operations on that file will fail with ENOENTRY reason, since all **lookup()/proc_lookup()** calls
will fail before given request even reaches fs port.

closed 2026-11-12 by damianloew