phoenix-rtos/phoenix-rtos-project #1400: `fifo` blocks when opened for reading and writing

adamdebek, 2025-09-09:
A `fifo` should only block on `open()` when opened for reading or writing and the corresponding other side is not yet open. Opening a `fifo` for both reading and writing (`O_RDWR` or `r+`) should not block.

Reproduction:
```c
#include <stdio.h>
#include <stdlib.h>
#include <sys/stat.h>

int main(void)
{
    FILE *f;

    if (mkfifo("fifo", 0666) < 0) {
        perror("mkfifo");
        exit(EXIT_FAILURE);
    }

    printf("Before open\n");

    f = fopen("fifo", "r+");
    if (!f) {
        perror("fopen");
        exit(EXIT_FAILURE);
    }

    printf("After open\n");

    return 0;
}
```

nalajcie, 2025-09-09:
Uhh, POSIX.1-2018 says that the result is undefined (https://pubs.opengroup.org/onlinepubs/9699919799/functions/open.html)

[image]

The same is also written in `man -s 7 fifo`:

[image]

I see that POSIX.1-2024 says that the `open` should return error if it's not supported so we could easily make our implementation conformant (https://pubs.opengroup.org/onlinepubs/9799919799/functions/open.html). Still, opening a fifo for RW should be avoided in portable code.

[image]

rmikielis, 2026-09-17:
For now posixsrv just skips sending msgRespond(), hanging user app indefinitely. Shall we fix it and respond with error, or I can delete it from foundation API issues list? @nalajcie

nalajcie, 2026-09-18:
Let's return an error - better user API. We might leave a TODO if we would like to support O_RDWR in the future.

closed 2026-09-29 by rmikielis